hathach/tinyusb #1984: ESP32S3 Composite HID Device with Generic IO, Keyboard and Mouse not connecting to PC

z-qh, 2023-03-25:
### Operating System

Linux

### Board

ESP32S3

### Firmware

IDF-5.0 and tinyUSB

### What happened ?

When I create a composite HID device with ESP32S3, I encounter a problem where the PC cannot connect to the device, regardless of whether the configuration descriptor exceeds 64 bytes or not. The composite device consists of a Generic Input Output Device, a Keyboard Device, and a Mouse Device.

### How to reproduce ?

composite generic input/output and keyboard desc:
const uint8_ t mine_ configuration_ descriptor[0x42] =
{
    /*Configuration Descriptor*/
    0x09,//1. Descriptor length
    0x02,//2. Descriptor Type Configuration Descriptor
    0x42,//3. Total length of configuration descriptor set=configuration descriptor+interface descriptor+HID class descriptor+endpoint descriptor
    0x00,//The low byte combination before is 0x0042
    0x02,//5. There are two interfaces in the current configuration=sending key values+receiving upper computer data
    0x01,//6. Identification of the current configuration
    0x00,//7. The currently configured string index value
    0xA0,//8. Bitwise configuration 7. Reserved bit 1 6. Bus power supply required bit 0 5. Remote wakeup support bit 1 4. - 0. All reserved bits are 0
    0x32,//9. The applied current is 2mA per unit

    /*Interface Descriptor - Keyboard*/
    0x09,//1. Descriptor length
    0x04,//2. Descriptor Type Interface Descriptor
    0x00,//3. Interface descriptor number
    0x00,//4. Interface descriptor Alternate interface number
    0x01,//5. The number of endpoints used by this interface uses 1 non 0 endpoint, only sending but not receiving, so 1 endpoint
    0x03,//6. Interface type HID interface
    0x01,//7. Interface subtype supports boot
    0x01,//8. Interface protocol 01 Keyboard 02 Mouse
    0x00,//9. The string index value of the interface is not named

    /*HID Class Descriptor*/
    0x09,//1. Descriptor length
    0x21,//2. Descriptor type HID class descriptor
    0x00,//3.HID version number
    0x01,//The low byte combination before 0x0100=1.0
    0x21,//5. Device Country Code American Keyboard
    0x01,//6. The number of subordinate descriptors contained contains a keyboard's report descriptor
    0x22,//7. Lower level descriptor type report descriptor
    0x41,//8. The length of the lower level descriptor is not fixed
    0x00,//Low byte first

    /*Endpoint Descriptor*/
    0x07,//1. Descriptor length
    0x05,//2. Descriptor Type Endpoint Descriptor
    0x81,//3. Describing by bit The lower 4 bits describe the number of endpoints 0000001 The 4th to 6th bits are reserved as 0 The 7th bit indicates the input and output direction (host viewing angle) 0 Output 1 Input
    0x03,//4. Bitwise description Low 2 bits indicate transmission type=00 Control ()+01 Synchronization (audio and video)+10 Batch transmission (printer)+11 Interrupt transmission (mouse and key) Other bits are useful during synchronization Currently, they are kept at 0
    0x40,//5. Two-byte bitwise description 0-10 bits indicate the maximum packet size 11-12 bits are valid under synchronization and interruption
    0x00,//6. Low bytes are only transmitted once in the first 00. 01 Retransmission once. 10 Retransmission twice. 11 Reservation
    0x0a,//7. The time interval for the host to send requests is 10ms

    /*Interface Descriptor - Custom HID Device*/
    0x09,//1. Descriptor length
    0x04,//2. Descriptor Type Interface Descriptor
    0x01,//3. Interface descriptor number
    0x00,//4. Interface descriptor Alternate interface number
    0x02,//5. The number of endpoints used by this interface uses two non zero endpoints for both sending and receiving
    0x03,//6. Interface type HID interface
    0x00,//7. Interface subtype does not support boot
    0x00,//8. Interface subtype common 0
    0x00,//9. The string index value of the interface is not named

    /*HID Class Descriptor*/
    0x09,//1. Descriptor length
    0x21,//2. Descriptor type HID class descriptor
    0x00,//3.HID version number
    0x01,//The low byte combination before 0x0100=1.0
    0x00,//5. Device country/region code customization
    0x01,//6. The number of subordinate descriptors included includes a custom report descriptor
    0x22,//7. Lower level descriptor type report descriptor
    0x23,//8. The length of the lower level descriptor is not fixed
    0x00,//Low byte first

    /*Endpoint Descriptor - Send*/
    0x07,//1. Descriptor length
    0x05,//2. Descriptor Type Endpoint Descriptor
    0x82,//3. Endpoint number 2. Host viewing angle is input
    0x03,//4. Interrupt transmission
    0x40,//5. Maximum packet length not retransmitted
    0x00,//Low byte first
    0x0a,//7. Host sends request interval of 10ms

    /*Endpoint Descriptor - Receive*/
    0x07,//1. Descriptor length
    0x05,//2. Descriptor Type Endpoint Descriptor
    0x02,//3. Endpoint number 2. Host viewing angle is output
    0x03,//4. Interrupt transmission
    0x40,//5. Maximum packet length not retransmitted
    0x00,//Low byte first
    0x0a//7. Host sends request interval of 10ms
};

### Screenshots

Capture screenshots of composite generic input/output and keyboard：
[https://great.wzznft.com/i/2023/03/25/lkgdvz.png](url)

Capture screenshots of using generic input/output separately:
[https://great.wzznft.com/i/2023/03/25/lkgm9t.png](url)

Capture screenshots of generic input and keyboard, delete the output section to save bytes:
[https://great.wzznft.com/i/2023/03/25/lkgqu4.png](url)

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Ryzee119, 2023-03-25:
`
/*HID Class Descriptor*/
0x09,//1. Descriptor length
`

Is that length right?

hathach, 2023-03-28:
what is your tinyusb log with CFG_TUSB_DEBUG = 2

z-qh, 2023-03-28:
> 

Please forgive my ignorance for not giving this debug message in the first time.
This is a debug message about "compound IO and keyboard device", failed：
```
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 18 00 00 00 00 00
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 FF 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes ...
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 2 bytes ...
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 03 03 09 04 FF 00
  Get Descriptor String[3]
  Queue EP 80 with 14 bytes ...
USBD Xfer Complete on EP 80 with 14 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 03 00 00 FF 00
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes ...
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 FF 00
  Get Descriptor String[2]
  Queue EP 80 with 34 bytes ...
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 06 00 00 0A 00
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 09 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 9 bytes ...
[... 257 lines cut ...]
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 01 03 09 04 FF 00
  Get Descriptor String[1]
  Queue EP 80 with 36 bytes ...
USBD Xfer Complete on EP 80 with 36 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 FF 00
  Get Descriptor String[2]
  Queue EP 80 with 34 bytes ...
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 0A 02
  Get Descriptor String[2]
  Queue EP 80 with 34 bytes ...
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

```

z-qh, 2023-03-28:
> what is your tinyusb log with CFG_TUSB_DEBUG = 2

After you prompted me to enable DEBUG, I used it to find a fatal error, one of the Settings was not set correctly, which caused me the above problem. I am very sorry that I did not check in place, and I am very sorry for the trouble caused to everyone. Now allow me to close the question.

closed 2023-03-29 by hathach