hathach/tinyusb #1990: SImple HID parser for device

hathach, 2023-03-29:
### Related area

device stack, hid

### Hardware specification

all

### Is your feature request related to a problem?

without parser, device hid driver does not know if report id is part of the report or not and sometime may present data especially output report incorrectly ID = invalid + whole report. This may confuse application if not aware of the issue.

### Describe the solution you'd like

simple parser enough to detect report ID present.

@dhalbert 

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

sanjay900, 2023-04-02:
If possible it would be nice to know what report IDs exist too, looking to see if a report descriptor supports a specific feature request or not

hathach, 2023-04-04:
usbd will only parser infomation that it needs, others that involed application code should be done by application.

sanjay900, 2023-04-04:
Yeah makes sense, i ended up writing my own simple parser for grabbing what i needed in the end.

SelvinPL, 2023-04-14:
@sanjay900 
https://gist.github.com/SelvinPL/99fd9af4566e759b6553e912b6a163f9
Works with most USB gamepads (nintendo switch pro not included but tested with PS4 and generic pads over USB)

sanjay900, 2023-04-14:
you know, i totally missed that too, but this will help me a bit thank you