hathach/tinyusb #1997: Data Seq Error when connecting HID device through USB hub

Xanvarok, 2023-04-03:
### Operating System

Linux

### Board

Raspberry Pi Pico

### Firmware

/examples/host/cdc_msc_hid

### What happened ?

When connecting a USB keyboard (or mouse) to the Pico via a USB hub, the process fails when trying to setup the keyboard (or mouse) throwing a Data Seq Error.

### How to reproduce ?

1. Apply power to Pico
2. Connect USB hub to Pico
3. Connect a USB keyboard (or mouse) to the USB hub
Or
1. Connect USB hub to Pico
2. Connect a USB keyboard (or mouse) to the USB hub
3. Apply power to Pico

### Debug Log as txt file

[TinyUSB_Data_Seq_Error.txt](https://github.com/hathach/tinyusb/files/11133843/TinyUSB_Data_Seq_Error.txt)

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-04-04:
do you have other USB hub to test with, just to make sure it is not as generic issue ? I tested previously fine with my hub.

Xanvarok, 2023-04-06:
Unfortunately I do not currently. 

The hub does work with both a raspberry pi 4 and with my desktop; however, it does look to be based around a HS8836 which other folks online seem to have issues with in other situations.

I'll see if I can get my hands on another hub to test with.

hathach, 2023-04-06:
when you are able to test with other hub and can narrow the issue to the specific chipset, please detail your hub make and model as well as link, picture etc.. If it is the case, I will try to see if I could find it locally here for testing.

Xanvarok, 2023-04-11:
So I managed to get my hands on a Sabrent HB-MCRM and everything seems to be working fine with that  ([log of working one here](https://github.com/hathach/tinyusb/files/11205540/TinyUSB_No_Error_With_Sabrent.txt)).

The Sabrent one isn't mine (so I didn't tear it apart) but it seems to claim it's using a Genesys Logic controller... probably a GL850G.

The one that doesn't work is a no-name brand... looks like the controller is actually a Huasheng Electronics HS8836A which seems to be found in a variety of cheap Chinese no-name USB hubs. There are no identifying marks on the case and the PCB is very basic... only silk-screened with "HS8836A-2022-12-02". I'll see if I can track down where it came from if you're interested.

morio, 2023-05-30:
After #2016 merged, I'm getting Data Seq Error when LOG=3 which is the same error when LOG=0. So at least it is consistent. 
Here are the logs when I power up with the hub already plugged in:
[usb-hub-setup.txt](https://github.com/hathach/tinyusb/files/11602841/usb-hub-setup.txt)

and then after I plug in the B100 Logitech mouse into the hub:
(ignore `ALL: Resetting devices` lines, as that is part of my code)

[usb-hub-connect-mouse.txt](https://github.com/hathach/tinyusb/files/11602844/usb-hub-connect-mouse.txt)

If I plug a known working hub into the Data Seq error hub and plug the mouse into the working hub, the mouse works fine.
(RP2040->(Data Seq error hub)->(Known working hub)->(B100 Mouse) - works
(RP2040->(Data Seq error hub)->(B100 Mouse) - errors

HiFiPhile, 2026-11-09:
#4458 fixes the host-side "Data Seq Error" causes found on the bench (RP2040: `74f8cbc`, `d92777f`; RP2350: `964943e`), resends instead of panicking on one that remains (`9000f89`), and lets a low-speed HID behind a hub retry a missed transaction (`342c141`). Could you retest with #4458?