hathach/tinyusb #2041: Problem when sending exact multiple endpoints size with CDC

hathach, 2023-04-25:
### Operating System

Linux

### Board

pico

### Firmware

cdc_msc example with CFG_TUD_CDC_EP_BUFSIZE = 256

### What happened ?

send 256bytes, with exact 64 bytes each. DCD won't notify tinyusb until all 256 bytes are received even there is a huge delay between each 64-byte packet. 

### How to reproduce ?

run following script on PC

```python
from serial import Serial

serial = Serial('COM5', baudrate=4_000_000)

serial.write(b'\x46'*64) 
print(serial.readline())

```

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

n/a

### Screenshots

ref: https://github.com/micropython/micropython/issues/11253

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

tore-espressif, 2023-05-19:
@hathach isn't this correct and expected behavior? If the transfer did not finish with Zero Length Packet, the DCD layer should wait for more data.

Could we add a ZLP timeout? In case there is a transfer with length == MPS and there is no packet in the ZLP timeout, the DCD would notify TinyUSB about new received data

hathach, 2023-05-22:
@tore-espressif indeed, it is specs on, and nothing wrong with current implementation.though it still causes some confusion for micropython user in the reference issue above. I was trying to see if we could find a way "**resolve**" this specially with CDC.

HiFiPhile, 2026-10-31:
Current master no longer waits for the whole buffer. By default (`CFG_TUD_CDC_RX_NEED_ZLP = 0`) CDC queues one packet per RX transfer, so each 64-byte packet reaches the application as it arrives. Multi-packet RX transfers that end on a short packet or ZLP are opt-in with `CFG_TUD_CDC_RX_NEED_ZLP = 1`, which needs a host that sends the ZLP. Closing; please reopen if you still see it on master.

closed 2026-10-31 by HiFiPhile