hathach/tinyusb #2060: __builtin_ctz function is not supported by IAR intrinsic functions:

abakosh, 2023-05-11:
### Operating System

Windows 10

### Board

RA family

### Firmware

src/portable/renesas/rusb2/hcd_rusb2.c

Also in other places under the portable directory.

### What happened ?

__builtin_ctz function is not supported by IAR intrinsic which is used in the following:
src/portable/renesas/rusb2/hcd_rusb2.c

I am not sure if it is available under a different name. However, I would like to hear your opinion regarding this.

Thank you.

### How to reproduce ?

Just by compiling in IAR workbench

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

 Error[Li005]: no definition for "__builtin_ctz" [referenced from $PROJDIR$\tinyusb\src\portable\renesas\rusb2\hcd_rusb2.o]   

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

manuelbl, 2023-05-13:
IAR should support the official ARM compiler language extension (ACLE) if the header `arm_acle.h` is included.

It does not contain `__ctz` (count trailing zeros). But `__rbit` and `__clz`. So you could try with:

```
#define __builtin_ctz(x) __clz(__rbit(x))
```

abakosh, 2023-05-13:
I defined it to the local function. I want to try the solution you have here. Thanks