hathach/tinyusb #2074: dcd_dwc2.c - reset_core() halts in infinite loop

MarekRyn, 2023-05-22:
### Operating System

Windows 10

### Board

Custom STM32H747BGT6

### Firmware

Custom firmware with use of TinyUSB lib. Affected file: src/portable/synopsys/dwc2/dcd_dwc2.c

### What happened ?

dcd_dwc2.c - function reset_core() halts in infinite loop after restart or power up of MCU. It happens at random times.

Present code:
static void reset_core(dwc2_regs_t * dwc2)
{
  // reset core
  dwc2->grstctl |= GRSTCTL_CSRST;

  // wait for reset bit is cleared
  // TODO version 4.20a should wait for RESET DONE mask
  while (dwc2->grstctl & GRSTCTL_CSRST) { }     // <<---- Software halts here

  // wait for AHB master IDLE
  while ( !(dwc2->grstctl & GRSTCTL_AHBIDL) ) { }

  // wait for device mode ?
}

Solution:
As per ST reference manual RM0399 page 2011: The software must also check that bit 31 in this register is set to 1 (AHB Master is Idle) __before__ starting any operation. 

Also in HAL original library mentioned check is carried out _before_ reset operation (see: function USB_CoreReset in stm32h7xx_ll_usb.c)

Proposal for corrected code:

static void reset_core(dwc2_regs_t * dwc2)
{
  // wait for AHB master IDLE
  while ( !(dwc2->grstctl & GRSTCTL_AHBIDL) ) { }

  // reset core
  dwc2->grstctl |= GRSTCTL_CSRST;

  // wait for reset bit is cleared
  // TODO version 4.20a should wait for RESET DONE mask
  while (dwc2->grstctl & GRSTCTL_CSRST) { }

  // wait for device mode ?
}

### How to reproduce ?

Error happens randomly during restart or powering up of uC.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

N/A

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Zychon, 2025-01-17:
Your post helps a lot, thanks.
I have the exact same problem with STM32F723IE MCU.
Custom firmeware, board and using the internal hs phy of the f723.
Your solution does not work on my board, but the problem does not happen when I add a delay at the end of dwc2_phy_init(). 

### TinyUsb Configuration
 - NO OS
 - CDC + MSC
 - USB HS
 - DMA
### Personal solution in dwc2_stm32.h
```c
static inline void dwc2_phy_init(dwc2_regs_t* dwc2, uint8_t hs_phy_type) {
  if (hs_phy_type == GHWCFG2_HSPHY_NOT_SUPPORTED) {
    // Enable on-chip FS PHY
    dwc2->stm32_gccfg |= STM32_GCCFG_PWRDWN;
      
      .
      .
      .

      // Control the tuning interface of the High Speed PHY
      // Use magic value (USB_HS_PHYC_TUNE_VALUE) from ST driver for F7
      USB_HS_PHYC->USB_HS_PHYC_TUNE |= 0x00000F13U;

      // Enable PLL internal PHY
      USB_HS_PHYC->USB_HS_PHYC_PLL |= USB_HS_PHYC_PLL_PLLEN;
      #else

      #endif
    }
     HAL_Delay(100);
  }

}
```
Found this in the Reference manual of STM32F72xxx and STM32F73xxx on site 1216
[image]

ST also add a delay after hs_phy init, maybe it is the solution :) found in stm32f7xx_ll_usb.c
[image]

Dolphindalt, 2025-09-05:
Seeing a similar issue on the STM32WBA65RI.

closed 2025-09-10 by hathach

Dolphindalt, 2025-09-10:
Was able to overcome the issue on the STM32WBA65RI by using a shadow register for the dwc2 version ID before resetting the device.

PR #3126 (2025-09-10): [stm32] Wait until the PHYC PLL is stable
**Describe the PR**

Thank you for this awesome project, very enjoyable to work with!

I've been integrating TinyUSB using custom firmware on the STM32F723E-DISCOVERY board and got stuck in `reset_core` when initializing the HS port. Waiting 2ms after PLL enablement fixes this issue.

**Additional context**

The solution for this was already discovered in https://github.com/hathach/tinyusb/issues/2074#issuecomment-2597894247, where the CubeHAL waits 2ms, since there's no USB_HS_PHYC_PLLRDY bit in the USB_HS_PHYC peripheral

Closes #2074.