hathach/tinyusb #2090: RP2040: Dual UAC2 Audio devices caused tud_audio_rx_done_pre_read_cb callbacks on audio_n_write

HiFiPhile, 2023-06-01:
### Discussed in https://github.com/hathach/tinyusb/discussions/2083

<div type='discussions-op-text'>

<sup>Originally posted by **mkarliner** May 26, 2023</sup>
### Operating System

MacOS

### Board

Raspberry Pi Pico

### Firmware

Based on UAC2 headset example, extended to have 2 UAC2_Headset devices
The headset example works as expected.

### What happened ?

I have two UAC2 devices running.
- A device called 'L' acts as a microphone, sending data from an electret microphone (using the RPi microphone library)
up to the Mac using audio_n_write.
- The Mac runs sox, consuming the mic data and sending it to a wav file. Note the Mac is not outputting to either of the device's speaker channel.
-  The tud_audio_rx_done_pre_read_cb callback is triggered, despite no data being sent to the speaker channels.

### How to reproduce ?

On Mac : sox -r 44100 -t coreaudio "VM Loop" -c 2 test.wav
tud_audio_rx_done_pre_read_cb is triggered.

zipfile attached. evidence.c is a redacted version of main.c for readability.

interface enum and endpoint defs are as follows
num
{
  
  ITF_NUM_AUDIO1_CONTROL = 0,
  ITF_NUM_AUDIO1_STREAMING_SPK,
  ITF_NUM_AUDIO1_STREAMING_MIC,
  //   ITF_NUM_CDC,
  // ITF_CDC_DATA,
  // ITF_NUM_AUDIO2_CONTROL ,
  ITF_NUM_AUDIO2_CONTROL,
  ITF_NUM_AUDIO2_STREAMING_SPK,
  ITF_NUM_AUDIO2_STREAMING_MIC,

  ITF_NUM_TOTAL
};

#define EPNUM_AUDIO1_IN 0x01
#define EPNUM_AUDIO1_OUT 0x01
#define EPNUM_AUDIO2_IN 0x02
#define EPNUM_AUDIO2_OUT 0x02
[evidence.zip](https://github.com/hathach/tinyusb/files/11574661/evidence.zip)

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

WWW2
  Queue EP 02 with 388 bytes ...
USBD Xfer Complete on EP 82 with 0 bytes
  AUDIO xfer callback
  Queue EP 82 with 0 bytes ...
USBD Xfer Complete on EP 02 with 176 bytes
  AUDIO xfer callback
WWW2
  Queue EP 02 with 388 bytes ...
USBD Xfer Complete on EP 82 with 0 bytes
  AUDIO xfer callback
  Queue EP 82 with 0 bytes ...

USBD Setup Received 01 0B 00 00 04 00 00 00
  Set Interface
  AUDIO control request
  Set itf: 4 - alt: 0
  CLOSING Endpoint: 0x02
Set interface 4 alt 0
  Queue EP 80 with 0 bytes ...
USBD Xfer Complete on EP 82 with 0 bytes
  AUDIO xfer callback
  Queue EP 82 with 0 bytes ...
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 01 0B 00 00 05 00 00 00
  Set Interface
  AUDIO control request
  Set itf: 5 - alt: 0
  CLOSING Endpoint: 0x82
Set interface 5 alt 0
  Queue EP 80 with 0 bytes ...
USBD Xfer Complete on EP 80 with 0 bytes

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.</div>

mkarliner, 2023-06-07:
Is there any more information I can give to help with resolving this issue?

HiFiPhile, 2026-10-31:
`tud_audio_rx_done_pre_read_cb()` no longer exists: the audio class has been reworked since 2023 and its callbacks changed. Closing as obsolete; please open a new issue against current master if two UAC2 functions still misbehave.

closed 2026-10-31 by HiFiPhile