hathach/tinyusb #2114: [STM32] CFG_TUSB_RHPORT1_MODE is not set in dcd_init

hitech95, 2023-06-16:
### Operating System

Windows 10

### Board

stm32f401blackpill

### Firmware

Custom firmware STM32Arduino and Platformio based.
Main contains the additional required code to set he GPIO pinmux, and VSENSE is disabled via:
```c
#include <Arduino.h>
#include "tusb.h"

// Pinmux for USB not exposed by core variants !
const PinMap PinMap_USB_OTG_FS[] = {
    {PA_11, USB_OTG_FS, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF10_OTG_FS)}, // USB_OTG_FS_DM
    {PA_12, USB_OTG_FS, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF10_OTG_FS)}, // USB_OTG_FS_DP
    {NC, NP, 0}
};

static void setupTinyUSB()
{
  const PinMap *map = NULL;

  // set pin mapping (using core library)
  map = PinMap_USB_OTG_FS;
  while (map->pin != NC)
  {
    pin_function(map->pin, map->function);
    map++;
  }
  
  /* Enable USB FS Clock */
  __HAL_RCC_USB_OTG_FS_CLK_ENABLE();
  
  // Blackpill doesn't use VBUS sense (B device) explicitly disable it
  USB_OTG_FS->GCCFG |= USB_OTG_GCCFG_NOVBUSSENS;
  USB_OTG_FS->GCCFG &= ~USB_OTG_GCCFG_VBUSBSEN;
  USB_OTG_FS->GCCFG &= ~USB_OTG_GCCFG_VBUSASEN;
  
  tud_init(BOARD_TUD_RHPORT);
  }
  
void setup()
{
  setupTinyUSB();
}

void loop()
{
  tud_task(); // tinyusb device task
 }
  ```
  I have defined the inlude path for the tusb_config.h file and the CFG_TUSB_MCU=OPT_MCU_STM32F4 via the platformio.ini file
  ```ini
  [env:blackpill_f401cc]
platform = ststm32
board = blackpill_f401cc
framework = arduino
upload_protocol = dfu
build_flags = 
    -D CFG_TUSB_MCU=OPT_MCU_STM32F4
    -Iinclude/
monitor_dtr = 1
```

### What happened ?

`CFG_TUSB_RHPORT1_MODE` is not defined and I cannot find how it should be defined.

Error is thrown due to lack of definition:
```bash
lib\tinyusb\src\portable\st\synopsys\dcd_synopsys.c: In function 'dcd_init':
Compiling .pio\build\blackpill_f401cc\FrameworkArduino\VirtIOSerial.cpp.o
lib\tinyusb\src\portable\st\synopsys\dcd_synopsys.c:528:25: error: 'CFG_TUSB_RHPORT1_MODE' undeclared (first use in this function); did you mean 'TUH_RHPORT_MODE'?
  528 |   if (!(rhport == 1 && (CFG_TUSB_RHPORT1_MODE & OPT_MODE_HIGH_SPEED))) usb_otg->GCCFG |= USB_OTG_GCCFG_PWRDWN;
      |                         ^~~~~~~~~~~~~~~~~~~~~
      |                         TUH_RHPORT_MODE
lib\tinyusb\src\portable\st\synopsys\dcd_synopsys.c:528:25: note: each undeclared identifier is reported only once for each function it appears in
*** [.pio\build\blackpill_f401cc\lib7f8\tinyusb\portable\st\synopsys\dcd_synopsys.c.o] Error 1
============================================================================ [FAILED] Took 6.84 seconds ============================================================================
```

There is a bug #1945 about the same issue but is not clear how I should use dwc2.

I suppose that the port definition changed and the `st/synaptic` driver is still using the old one. while it is not imported anymore it seems that platformio builds all .c file resulting in the error.

### How to reproduce ?

1. download included project
2. open project in vscode/platformio ide
3. build
[TestUSB.zip](https://github.com/hathach/tinyusb/files/11772262/TestUSB.zip)
[log_tinyusb.txt](https://github.com/hathach/tinyusb/files/11772182/log_tinyusb.txt)

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

none

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-06-16:
> There is a bug https://github.com/hathach/tinyusb/issues/1945 about the same issue but is not clear how I should use dwc2.

add src/portable/synopsys/dwc2/dcd_dwc2.c and remove the other st driver.

closed 2023-06-16 by hathach

hitech95, 2023-06-16:
> add src/portable/synopsys/dwc2/dcd_dwc2.c and remove the other st driver.

so the only way is to to modify the imported library? (git submodule)