hathach/tinyusb #2123: USB stick does not work unless plugged in at program start

DRNadler, 2023-06-24:
### Operating System

Windows 10

### Board

imxRT1024 custom (USB same as EVK board)

### Firmware

tinyUSB main as of noon EST 24-June-2023.

arm-none-eabi-g++ (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
Copyright (C) 2020 Free Software Foundation, Inc.

Application is essentially msc_file_explorer example (FreeRTOS).

### What happened ?

If the application is started with the USB stick already plugged in, it works fine.
```
A MassStorage device is mounted
Disk: Generic  Flash Disk       rev 8.07
Disk Size: 1911 MB
USB label: 2GB_SCRATCH
```
If the application is started and then the USB stick plugged in, fails as follows (LOG=2).
Similar behavior if the stick is unplugged then plugged in again:
```
USBH init on controller 0
sizeof(usbh_device_t) = 86
sizeof(hcd_event_t) = 12
sizeof(_ctrl_xfer) = 24
sizeof(tuh_xfer_t) = 24
sizeof(tu_fifo_t) = 20
sizeof(tu_edpt_stream_t) = 112
MSC init
```
Now plug in USB stick...
```
[0:] USBH DEVICE ATTACH
Full Speed
[0:0] Open EP0 with Size = 8
Get 8 byte of Device Descriptor
[0:0] Get Descriptor: 80 06 00 01 00 00 08 00
on EP 00 with 0 bytes: FAILED
[0:0] Control FAILED, xferred_bytes = 0

Enumeration attempt 1
[0:0] Get Descriptor: 80 06 00 01 00 00 08 00
on EP 00 with 0 bytes: FAILED
[0:0] Control FAILED, xferred_bytes = 0

Enumeration attempt 2
[0:0] Get Descriptor: 80 06 00 01 00 00 08 00
on EP 00 with 0 bytes: FAILED
[0:0] Control FAILED, xferred_bytes = 0

Enumeration attempt 3
[0:0] Get Descriptor: 80 06 00 01 00 00 08 00
on EP 00 with 0 bytes: FAILED
[0:0] Control FAILED, xferred_bytes = 0
```

### How to reproduce ?

Should be reproducible using msc_file_explorer on RT1024EVK board.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

See above.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-06-26:
If you have evk, please test it with evk. And absolutely need to test with stock example without freeRTOS, and 99% same firmware are still 2 different firmware.

DRNadler, 2023-08-03:
Apologies I won't be able to look at this for some weeks...

DRNadler, 2024-01-04:
Update as I've had a chance to further test this...

Oddly, plugging in a USB stick with application does work with a new faster USB stick (SanDisk Extreme PRO 128GB), but fails with older USB sticks. Looks like a timing error or maybe another race condition. I'll report more when I've had time to further debug...

DRNadler, 2024-05-13:
Further info: This behavior is caused by a very slow old USB stick. I had to increase the MSD read timeout for this stick to work with the uTasker USB stack. Unfortunately there are lots of sticks like this around... How do I increase the MSD read time-out value for tinyUSB?
Thanks!