hathach/tinyusb #2158: MIDI as part of a Composite Device

PaulFortescue, 2023-07-20:
### Operating System

Windows 11

### Board

Raspberry Pi Pico

### Firmware

examples/device/msc is the base of this.

I can incorporate msc/cdc/hid into a composite device using the examples provided and it works perfectly.

I can use MIDI as a standalone device using the example provided and it works perfectly.

I cannot, using any permutation or combination of settings, make MIDI work concurrently with any other device or devices. Windows rejects the device with a (not very useful) code 10 meaning "something is wrong".

However, I deduce from the internet that it IS possible to incorporate MIDI as part of a composite device.

### What happened ?

I am only guessing that there is a bug somewhere that prevents me using TinyUSB to do this?

I've spent 3 days solid on it now so I'm hoping for an answer, even if it's "no"!

PS many, many thanks for the TinyUSB library!

### How to reproduce ?

Working with the "//" lines in, not with them out:

enum
{
	ITF_NUM_CDC = 0,
	ITF_NUM_CDC_DATA,
	ITF_NUM_MSC,
//	ITF_NUM_MIDI,
//	ITF_NUM_MIDI_STREAMING,
	ITF_NUM_TOTAL
};

#define CONFIG_TOTAL_LEN    (TUD_CONFIG_DESC_LEN + TUD_CDC_DESC_LEN + TUD_MSC_DESC_LEN) 
// + TUD_MIDI_DESC_LEN 

uint8_t const desc_fs_configuration[] =
{
	// Config number, interface count, string index, total length, attribute, power in mA
	TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 0, CONFIG_TOTAL_LEN, 0x00, 100),

	// Interface number, string index, EP notification address and size, EP data address (out, in) and size.
	TUD_CDC_DESCRIPTOR(ITF_NUM_CDC, 4, EPNUM_CDC_NOTIF, 8, EPNUM_CDC_OUT, EPNUM_CDC_IN, 64),

	// Interface number, string index, EP Out & EP In address, EP size
	TUD_MSC_DESCRIPTOR(ITF_NUM_MSC, 5, EPNUM_MSC_OUT, EPNUM_MSC_IN, 64),

//	TUD_MIDI_DESCRIPTOR(ITF_NUM_MIDI, 0, EPNUM_MIDI, 0x80 | EPNUM_MIDI, 64),

};

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

//

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-07-21:
midi work just fine with other class, try out arduino example here 
- https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/MIDI/midi_test

Arduino always keep the cdc therefore this is cdc + midi. You probably get the descriptor incorrectly. You can troubleshoot this by enabling debug log and/or dumping configuration and use online parse tool to spot any error.

Btw your how-to-reproduce is not reproducible, it is best to post a link to your example repo.

PaulFortescue, 2023-07-21:
Hello Mr Thach and thank you very much for your quick reply!

I'm going to attempt to put my code on here, but I've never used GitHub except for looking at code, so bear with me.

I've solved my problem by having two different code sets which I will use one at a time, as I don't actually really need composite (I just need the same port to be either MIDI or CDC depending on a switch).

PaulFortescue, 2023-07-21:
I've posted all the code at https://github.com/PaulFortescue/Miscellaneous

(I think)

If anyone wants to put me right I'm listening. I probably don't know what I'm doing, so please don't shout at me.

All I know is that the code WILL work if I take MIDI out, or WILL work if I ONLY leave MIDI in. See original post for details.