hathach/tinyusb #2161: xHCI controller support

hathach, 2023-07-21:
### Related area

Host Driver (HCD)

### Hardware specification

pi400

### Is your feature request related to a problem?

Support xHCI host controller.

**motivation**

pi400 has an keyboard connected to xHCI controller via Holtek MCU. It would be nice to have ported so that we can have circuitpython as a PC. Note: xHCI is connected to bcm cpu via pci xpress which can be a bit of challange, we will see.  

```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04d9:0007 Holtek Semiconductor, Inc. Raspberry Pi Internal Keyboard
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
```

### Describe the solution you'd like

pi400 running baremetal example working with its integrated keyboard

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

cleverca22, 2023-07-27:
an added complication to beware of

the vl805 pcie<->usb3 controller, normally has its own firmware in dedicated SPI flash, and once you configure the xhci scratch space, it will (ab)use the scratch space like swap, to store its own firmware/state

but RPF later found a way to save costs, if that SPI chip is omitted, thats 1 less part on the board

the 0x00030058/RPI_FIRMWARE_NOTIFY_XHCI_RESET mailbox call must be sent from the arm kernel to the VPU firmware, i think any time the pcie bus is reset
and that will side-load a bootloader into the VL805, along with pointers to a firmware image, allowing the VL805 to function without the dedicated SPI chip

the pi400 has always had that SPI chip missing
the pi4 started out with it, so different revisions will behave differently if you omit this

hathach, 2023-07-29:
Thank you for your info, that is definitely helpful info.