hathach/tinyusb #2166: EHCI QHD reuses QTD on wrong endpoint

tannewt, 2023-07-22:
### Operating System

Linux

### Board

iMX RT 1060 evk with usb drive

### Firmware

On this branch: https://github.com/tannewt/circuitpython/tree/three_small

CP code.py is:
```python
import usb.core
import storage
import struct
import time

device = None
while device is None:
    print("searching for devices")
    for device in usb.core.find(find_all=True):
        print("pid", hex(device.idProduct))
        print("vid", hex(device.idVendor))
        print("man", device.manufacturer)
        print("product", device.product)
        print("serial", device.serial_number)
        break
    if not device:
        time.sleep(5)

# USB defines
DIR_SHIFT = 7
DIR_OUT = 0 << DIR_SHIFT
DIR_IN = 1 << DIR_SHIFT

# No shift
REQ_RCPT_DEVICE = 0
REQ_RCPT_INTERFACE = 1
REQ_RCPT_ENDPOINT = 2
REQ_RCPT_OTHER = 3

REQ_TYPE_SHIFT = 5
REQ_TYPE_STANDARD = 0 << REQ_TYPE_SHIFT
REQ_TYPE_CLASS = 1 << REQ_TYPE_SHIFT
REQ_TYPE_VENDOR = 2 << REQ_TYPE_SHIFT
REQ_TYPE_INVALID = 3 << REQ_TYPE_SHIFT

REQ_GET_DESCRIPTOR = 6
REQ_SET_INTERFACE = 11

DESC_DEVICE = 0x01
DESC_CONFIGURATION = 0x02
DESC_STRING = 0x03
DESC_INTERFACE = 0x04
DESC_ENDPOINT = 0x05

# IOCTL values
CTRL_SYNC          = 0   # Complete pending write process (needed at FF_FS_READONLY == 0)
GET_SECTOR_COUNT   = 1   # Get media size (needed at FF_USE_MKFS == 1)
GET_SECTOR_SIZE    = 2   # Get sector size (needed at FF_MAX_SS != FF_MIN_SS)
GET_BLOCK_SIZE     = 3   # Get erase block size (needed at FF_USE_MKFS == 1)
CTRL_TRIM          = 4   # Inform device that the data on the block of sectors is no longer used (needed at FF_USE_TRIM == 1)
IOCTL_INIT         = 5
IOCTL_STATUS       = 6

# SCSI commands
SCSI_CMD_TEST_UNIT_READY              = 0x00 # The SCSI Test Unit Ready command is used to determine if a device is ready to transfer data (read/write), i.e. if a disk has spun up, if a tape is loaded and ready etc. The device does not perform a self-test operation.
SCSI_CMD_INQUIRY                      = 0x12 # The SCSI Inquiry command is used to obtain basic information from a target device.
[... 179 lines cut ...]
  ehci xfer error 02 on 1 result 1
  qhd 0x202013c0 qtd 0x20201c40 current 0x20201c40
  cb done dev 1 ep 02 on 0x60050419
   res 1 len 0
xfer dev 1 ep 81 to 0x60050419
ehci xfer to 81 on 1
  qhd 0x20201400 qtd 0x20201c40
  ehci xfer error 81 on 1 result 1
  qhd 0x20201400 qtd 0x20201c40 current 0x20201c40
  ehci xfer error 82 on 1 result 1
  qhd 0x202013c0 qtd 0x20201c40 current 0x20201c40
  cb done dev 1 ep 81 on 0x60050419
   res 1 len 0
  cb done dev 1 ep 82 on 0x0
   res 1 len 0
```

This is my own printf debugging. The second to last line shows that the done cb is NULL for ep 82, which I never initiated ("ehci xfer to"). This is right after an error on the correct endpoint. Both qhd's have the same qtd set as current. So, I suspect we aren't completely detaching a qtd from the qhd before reusing it.

Full log with tinyusb logging on too:
[imx1060_tinyusb_log.txt](https://github.com/hathach/tinyusb/files/12135227/imx1060_tinyusb_log.txt)

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-07-24:
hmm, I don't quite reproduce the issue, I got follow message from REPL and no TU_ASSERT breakpoint with debugger

```
You are in safe mode because:                                                        
CircuitPython core code crashed hard. Whoops!                                        
Fault detected by hardware.                                                          
Please file an issue with your program at https://github.com/adafruit/circuitpython/i
ssues.                                                                               
Press reset to exit safe mode.                                                       
                                                                                     
Press any key to enter the REPL. Use CTRL-D to reload.     
```

tannewt, 2023-07-24:
This is still happening for me. The drive is labeled DT101 G2 16GB. Here is a screenshot of the beagle capture:
[image]

The OUT transactions to EP 02 are failing but CP is continuing. The IN transactions to EP 81 also fail and when they do, TinyUSB is finding the EP 02 failures a second time. The overlay isn't reset out of the failure state when the error is handled. I think there needs to be a check there that `attached_qtd` matches the overlay's current qtd. The detach code also needs to clean the housekeeping values back out to ram, otherwise the invalidate in the subsequent ISR doesn't see the NULL set to `attached_qtd`.

tannewt, 2023-07-24:
Also, moving the housekeeping data out of the QHD will take more RAM but make cache management simpler and is probably worthwhile.

tannewt, 2023-07-24:
I think my original error was caused by sending the wrong value in "set configuration" (0 instead of 1). Maybe that could help reproduce it for you.

PR #2179 (2023-07-27): Enhance EHCI
**Describe the PR**
- Fix issue with hub status_change is not aligned
- EHCI use standard USB complete interrupt instead of custom chipidea async/period interrupt to be more compatible with other ehci implementation
- refactor usb complete & error isr processing, merge, update. Should probably fix #2166 . @tannewt please try it out when you have time.
closed 2023-07-27 by hathach