hathach/tinyusb #2177: host mode enumeration halts if configuration descriptor too long

cleverca22, 2023-07-24:
### Operating System

Linux

### Board

rpi2

### Firmware

custom https://github.com/librerpi/lk-overlay/tree/master/platform/bcm28xx/dwc2 + https://github.com/librerpi/lk-overlay/tree/master/lib/tinyusb

### What happened ?

`TU_ASSERT(total_len <= CFG_TUH_ENUMERATION_BUFSIZE, );`
during enumeration, tinyusb will check if a config descriptor fits into the available buffer space
a usb webcam has a config descriptor over 1024 bytes, but the buffer is only 512 bytes

due to this assert, tinyusb just grinds to a halt, and stops enumerating, when it could have just left the device unconfigured and move to the next usb port

### How to reproduce ?

connect something with a large config descriptor(a webcam for example) to a tinyusb host (any board/firmware should work)

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

```
process_enumeration() daddr=2 state=9
Get Configuration[0] Descriptor (9 bytes)
[0:2] Get Descriptor: 80 06 00 02 00 00 09 00 
 37.643664 [DWC2:hcd_setup_send:346]:   HOST0 SETUP 2.00 0xc4014d70/8 opep:0xc411418c
dumping channel 0
  HCCHAR0: 0x900008
    mps:8 ep:0 dir0 type:0 mc:1 addr:2
  HCSPLT0: 0x0
  HCINT0: 0x0
  HCINTMSK0: 0x7ff
  HCTSIZ0: 0x20000008
    size:8 packets:0 pid:1 ping:0
  HCDMA0: 0x0
 37.669686 [DWC2:dwc_check_interrupt:603]: hcd_event_xfer_complete(2, 0x0, 8, success, true)
 37.677843 [DWC2:dwc_irq:644]: irq time: 25849
on EP 00 with 8 bytes: OK
 37.792983 [DWC2:hcd_edpt_xfer:368]:    HOST0  <-   2.80 0xc4014d88/9 opep:0xc411418c type:0 pid:153
 38.801206 [DWC2:dwc_check_interrupt:500]: hcd_event_xfer_complete(2, 0x80, 9, success, true)
 38.807972 [DWC2:dwc_irq:644]: irq time: 6766
on EP 80 with 9 bytes: OK
[0:2] Control data:
  0000:  09 02 95 05 04 01 00 80 FA                       |.........|
 38.832633 [DWC2:hcd_edpt_xfer:376]:    HOST0  ->   2.00 0x0/0 opep:0xc411418c
dumping channel 0
  HCCHAR0: 0x900008
    mps:8 ep:0 dir0 type:0 mc:1 addr:2
  HCSPLT0: 0x0
  HCINT0: 0x0
  HCINTMSK0: 0x7ff
  HCTSIZ0: 0x0
    size:0 packets:0 pid:0 ping:0
  HCDMA0: 0x0
 38.855877 [DWC2:dwc_check_interrupt:603]: hcd_event_xfer_complete(2, 0x0, 0, success, true)
 38.864034 [DWC2:dwc_irq:644]: irq time: 25230
on EP 00 with 0 bytes: OK

process_enumeration() daddr=2 state=10
ENUM_GET_FULL_CONFIG_DESC
config size: 1429 vs 512
process_enumeration 1426: ASSERT FAILED
```
it successfully fetched the first 9 bytes of the config descriptor, noticed the size wont fit, and then gave up

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-07-25:
the fix is obvious, increase the configure buffer

cleverca22, 2023-07-25:
@hathach but is that really effective, when you dont want to support such a device, and you can never know how big the buffer will be?

you always run the risk of your host-mode device breaking, because the user plugged in something with a bigger descriptor then you planned for

hathach, 2023-07-25:
Ah i see, I skip the part where the host skip entirely rest of the hub. You are right, it should just skip device and continue though.