hathach/tinyusb #2188: External Custom USBH Class Driver Definitions

RobertDaleSmith, 2023-07-29:
### Related area

Currently I maintain my own branch of TinyUSB with an XInput driver implemented. It would be really nice if I could simply define these within my application though.

### Hardware specification

rp2040

### Is your feature request related to a problem?

Rather than having to maintain a unique branch of TinyUSB with my XInput driver, it would be really nice to have a mechanism to add it through the config within my application. I dug around but there doesnt seem to be a way to externally append a driver to the `usbh_class_drivers` array.

You can see an example of my XInput driver implemented within TinyUSB here:
https://github.com/RobertDaleSmith/tinyusb/blob/xinput/src/host/usbh.c#L136

Then you can see an example of those driver functions I use within my application here:
https://github.com/Ryzee119/tusb_xinput/blob/master/xinput_host.c

### Describe the solution you'd like

Does it make sense or is it planned to add the ability to expand the `usbh_class_drivers` from the application level? Maybe could be defined in `tusb_config.h`.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

cleverca22, 2023-07-29:
the general solution i see in other projects, is some linker trickery

basically, every driver that defines a `struct usbh_class_driver_t` object, has to put it into a special section with `__attribute__ ((section(".data.usb_class_drivers")))`

the linker script will then put all of those objects into one block of memory, and generate start/end symbols

`usbh.c` can then convert that back into the old style array
the major benefit of this method, is that you dont need to `#ifdef` every entry in the `usbh_class_drivers[]` and you can add drivers from any source file

RobertDaleSmith, 2023-07-30:
This looks like exactly what I need @cleverca22. 🙏

Do you happen to know off hand of any drivers using this approach that I can reference?

cleverca22, 2023-07-30:
https://github.com/littlekernel/lk/blob/30dc320054f70910e1c1ee40a6948ee99672acec/platform/pc/lapic.c#L59
https://github.com/littlekernel/lk/blob/master/top/include/lk/init.h#L72
```c
/*
 * Copyright (c) 2013 Travis Geiselbrecht
 *
 * Use of this source code is governed by a MIT-style
 * license that can be found in the LICENSE file or at
 * https://opensource.org/licenses/MIT
 */
#define LK_INIT_HOOK_FLAGS(_name, _hook, _level, _flags) \
    const struct lk_init_struct _init_struct_##_name __ALIGNED(sizeof(void *)) __SECTION("lk_init") = { \
        .level = _level, \
        .flags = _flags, \
        .hook = _hook, \
        .name = #_name, \
    };
#define LK_INIT_HOOK(_name, _hook, _level) \
    LK_INIT_HOOK_FLAGS(_name, _hook, _level, LK_INIT_FLAG_PRIMARY_CPU)

LK_INIT_HOOK(lapic, lapic_init_postvm, LK_INIT_LEVEL_VM);
```
this will create a struct, with a unique name, and put it into the lk_init section in the .o file

https://github.com/littlekernel/lk/blob/30dc320054f70910e1c1ee40a6948ee99672acec/top/init.c#L28-L29
```c
extern const struct lk_init_struct __start_lk_init __WEAK;
extern const struct lk_init_struct __stop_lk_init __WEAK;
```
this is then the range of addresses those objects occupy
```c
 for (const struct lk_init_struct *ptr = &__start_lk_init; ptr != &__stop_lk_init; ptr++) {
```
and you can just loop over the whole array

hathach, 2023-08-01:
thank you for your suggestion, the linker trick is nice but it requires linker script modification, which I don't really like since it is not partiular easy with some auto-linking IDE/project. External driver is possible with C code like devie stack with `usbd_app_driver_get_cb()`. We can do something similar to host stack.

cleverca22, 2023-08-02:
i also remembered another benefit of this trick
ipxe always compiles every driver, to prevent code-rot

and then at link time, it can select what the user actually wanted, so you can still get small binaries, but lesser used code still gets syntax/type-checked by the compiler

but yeah, you would need to at least inject the linker script into the linker args, i believe it can take multiple linker scripts

hathach, 2023-08-03:
@cleverca22 multiple linker script could work with gcc, but proriearty compiler such as IAR or ARMCC may not (not sure though). It is a great approach if we are working on a specific project/mcu and could dictate all compiler/linker progress. I wouldn't want to touch linker in the some platform such as esp-idf. Anyway, I still perfer to solve this with C than linker.

rppicomidi, 2023-08-13:
@hathach If you require a C only solution, then macros like these should do the trick.
In file `usbh_classdriver.h`
```
/*
   To define your own USB Host driver(s), invoke this macro one or more times in the definition of the
   CFG_TUH_EXTERNAL_DRIVERS macro in your local tuh_config.h file
*/
#define TUH_MAKE_EXTERNAL_DRIVER(ex_name, ex_init, ex_open, ex_set_config, ex_xfer_cb, close) \
    { \
      DRIVER_NAME(ex_name) \
      .init       = ex_init, \
      .open       = ex_open, \
      .set_config = ex_set_config, \
      .xfer_cb    = ex_xfer_cb, \
      .close      = ex_close \
    }, \
```
In file `usbh.c` in the definition of the `usbh_class_drivers[]` array, add
```
#ifdef CFG_TUH_EXTERNAL_DRIVERS
  CFG_TUH_EXTERNAL_DRIVERS
#endif
```
In the local project `tusb_config.h` file (example for the MIDI Host driver)
```
#define CFG_TUH_EXTERNAL_DRIVERS \
  TUH_MAKE_EXTERNAL_DRIVER("MIDI", midih_init, midih_open, midih_set_config, midih_xfer_cb, midih_close)
```

The file `hcd.h` used to define `CFG_TUH_ENDPONIT_MAX` based on what host drivers are supported. Now it
seems it is always 16 unless you define it otherwise in your configuration file. Do any other files need to be touched to implement this?

hathach, 2023-08-14:
thanks @rppicomidi , we will probably go with the same approach used by custom device class driver with callback for consitency. Also back then we once considered a smilar approach like your suggestion, but espressif team insist on dynamic drivers to only load them once enabled, which I think kind of making sense. I forgot the detail, but you could go through the long discussion in following issues
- https://github.com/hathach/tinyusb/issues/384
- https://github.com/hathach/tinyusb/pull/385

rppicomidi, 2023-08-14:
@hathach That method works too. Does your schedule permit you to work on this now? If not, I can work on a pull request.

hathach, 2023-08-14:
> @hathach That method works too. Does your schedule permit you to work on this now? If not, I can work on a pull request.

yes please, if possible, you could check the usbd implemenation for reference. I always have issue to catch up with PR. I still want to review and merge your midi host, but kept pushing by other works.

rppicomidi, 2023-08-14:
@hathach OK. Working on it. Will try to have a solution soon. If it works, I can withdraw my midihost pull request and just maintain the midihost driver out of tree.

rppicomidi, 2023-08-14:
@hathach The pull request should be consistent with what was done for the USB device driver side.

hathach, 2023-08-15:
closed by #2222, thank you @rppicomidi for your work. If posible, please keep the midihost pr open, I really want to merge that one, but haven't got time just yet due to crzay work previously. Will try to wrap it up soon enough. You can actual use external for developement since external has more priority than built-in, and we could sync once after a while or so. That would allow to more peoeple to work on the driver.

closed 2023-08-15 by hathach