hathach/tinyusb #2191: tinyusb support dma feature

jeffery88, 2023-07-31:
### Related area

DMA

### Hardware specification

synopsys dwc2

### Is your feature request related to a problem?

How can tinyusb support dma features when its operation seems to be all fifo based? It is recommended to add the implementation of dma to improve the transfer rate of usb

### Describe the solution you'd like

It's difficult when I want to use the dma function with the tinyusb.I hope tinyusb can support dma feature.And I would be happy to port synopsys's dma to tinyusb.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2023-07-31:
Hi,

We had some discussion regarding DMA support in the past. The difficulty is to have a portable abstraction layer to make DMA compatible with most MCUs, also DWC2 is used by many Cortex-M & RiSC-V MCU families. 

It's welcomed if you want to make a PR.

cleverca22, 2023-07-31:
one area ive seen makes the fifo/dma stuff a bit more obvious, in dwc host mode

the high-speed ping protocol is manual when in non-dma mode
so you dont write to the fifo, you just enable sending pings
and when the ping gets an ACK, then you write to the fifo, and send the actual OUT

but in dma mode, you configure the DMA addr and enable pinging
it will ping, but not do dma
once it gets an ACK from the device, it will fill its own FIFO with DMA, and then perform the OUT

so at the end, its still driven by all of the FIFO's, but its just using DMA to fill them at the right time

HiFiPhile, 2024-04-07:
There is an on going PR : [#2576](https://github.com/hathach/tinyusb/pull/2576)

hathach, 2024-09-25:
implemented by #2576 need CFG_TUD_DWC2_DMA = 1

closed 2024-09-25 by hathach