hathach/tinyusb #2195: Connecting a hub to a hub a second time assert faults on RP2040 Pico

Slamy, 2023-07-31:
### Operating System

Linux

### Board

RP2040 Pico

### Firmware

examples/host/cdc_msc_hid/src

With tusb_config.h:
`#define CFG_TUH_HUB                 2 // number of supported hubs`

Compiled with 
`cmake -DFAMILY=rp2040 .. -DPICO_SDK_FETCH_FROM_GIT=1 -DLOGGER=RTT -DCMAKE_BUILD_TYPE=Debug`

### What happened ?

Output on RTT:
assertion "!ep->active" failed: file "/home/andre/GIT/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c", line 602, function: hcd_setup_send

### How to reproduce ?

Two ways possible.
First:
Having OTG USB Hub connected to Pico board. Connect second USB Hub in USB A port of OTG hub.
Wait 1 second.
Remove second HUB
Wait 1 second.
Connect second HUB again.
Assertion fails

Second case.
Start with OTG USB HUB connected to Pico board with second HUB already connected to first.
Wait 1 second.
Remove second HUB
Wait 1 second.
Connect second HUB again.
Assertion fails

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Recompiled with
`cmake -DFAMILY=rp2040 .. -DPICO_SDK_FETCH_FROM_GIT=1 -DLOGGER=RTT -DCMAKE_BUILD_TYPE=Debug -DLOG=2 ..`

```
HUB Clear Feature: PORT_CONNECTION_CHANGE, addr = 8 port = 4
[0:8] Class Request: 23 01 10 00 04 00 00 00 
Transfer complete
on EP 00 with 8 bytes: OK
Buffer complete
Transfer complete
on EP 80 with 0 bytes: OK

[0:8:4] USBH DEVICE REMOVED
Device unplugged address = 9
  is a HUB device 9
  HUB close addr = 9
  Queue EP 81 with 1 bytes ... 
OK
[0:9:0] USBH DEVICE REMOVED
  Queue EP 00 with 1 bytes ... 
OK
Buffer complete
on EP 81 with 1 bytes: OK
HUB xfer callback
  Hub Status Change = 0x10
HUB Get Port Status: addr = 8 port = 4
[0:8] Class Request: A3 00 00 00 04 00 04 00 
assertion "!ep->active" failed: file "/home/andre/GIT/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c", line 602, function: hcd_setup_send
RTT:0>
```

The stack
```
#0  _exit (status=status@entry=1) at /home/andre/GIT/tinyusb/examples/host/cdc_msc_hid/build/_deps/pico_sdk-src/src/rp2_common/pico_runtime/runtime.c:187
#1  0x1000265c in __assert_func (file=file@entry=0x1000c288 "/home/andre/GIT/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c", line=line@entry=602, func=func@entry=0x1000c428 <__func__.0> "hcd_setup_send", failedexpr=failedexpr@entry=0x1000c400 "!ep->active") at /home/andre/GIT/tinyusb/examples/host/cdc_msc_hid/build/_deps/pico_sdk-src/src/rp2_common/pico_runtime/runtime.c:271
#2  0x1000581e in hcd_setup_send (rhport=rhport@entry=0 '\000', dev_addr=dev_addr@entry=8 '\b', setup_packet=setup_packet@entry=0x2000173c <_ctrl_xfer> "\243") at /home/andre/GIT/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c:602
#3  0x1000692a in tuh_control_xfer (xfer=xfer@entry=0x20041f78) at /home/andre/GIT/tinyusb/src/host/usbh.c:569
#4  0x10007060 in hub_port_get_status (hub_addr=hub_addr@entry=8 '\b', hub_port=<optimized out>, resp=resp@entry=0x2000308c <hub_data+8>, complete_cb=complete_cb@entry=0x10007175 <hub_port_get_status_complete>, user_data=user_data@entry=0) at /home/andre/GIT/tinyusb/src/host/hub.c:178
#5  0x10007354 in hub_xfer_cb (dev_addr=<optimized out>, ep_addr=<optimized out>, result=<optimized out>, xferred_bytes=<optimized out>) at /home/andre/GIT/tinyusb/src/host/hub.c:358
#6  0x10006826 in tuh_task_ext (timeout_ms=timeout_ms@entry=4294967295, in_isr=in_isr@entry=false) at /home/andre/GIT/tinyusb/src/host/usbh.c:494
#7  0x100007cc in tuh_task () at /home/andre/GIT/tinyusb/src/host/usbh.h:125
#8  main () at /home/andre/GIT/tinyusb/examples/host/cdc_msc_hid/src/main.c:54
```

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Slamy, 2023-07-31:
I nearly forgot to tell you that I applied git bisect already.
The culprit is 2c48050993777c586210af503504e612f31228d8.
Before that, the systems seems to be fine with 2 Hubs.

HiFiPhile, 2026-11-09:
The `!ep->active` assert is gone, and #4458 reworks device close (`8ca1071`) and the EPX scheduling around unplugs. Could you retest the hub-in-hub replug with #4458?