hathach/tinyusb #2224: Support for Bouffalo BL616 EHCI host

harbaum, 2023-08-15:
### Related area

Run complex (hub/multidevice)  setups connected to the BL616 host

### Hardware specification

BouffaloLabs BL616 

### Is your feature request related to a problem?

I recently stumbled upon the BouffaloLabs BL616 which is a cheap Risc-V microcontroller coming with WiFi and Bluetooth and some other interesting peripherals incl. a EHCI controller. The BL616 seems to target the same audience as the ESP32 and is roughly comparable.

This BouffaloLab SDK comes with cherryUSB but host mode only partially works. But I got some mass storage and HID examples to partially run.

The BL616 is available in asian online markets on boards like the SiPeed M0S dock and the AI Thinker AI M62 12F.

However, the USB host examples are not very reliable and e.g. I see data corruption in bulk out / msc write  transfers. I am not sure whether these are hardware or software issues. On the plus side I can read 8 MBytes/s from a USB MSC device which sounds rather promising.

The fact that the BL616 uses a EHCI controller suggests that it might be a minor task to use the tinyusb/ehci host driver.
 

### Describe the solution you'd like

I would like to run tinyusb ehci host on the BL616 and control a hub/kbd/mouse/gamepad/memory-stick setup.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-08-16:
The mcu looks interesting, unfortunately I have no plan to support this anytime soon yet.

harbaum, 2023-08-16:
Thanks for the fast reply, anyway.

I might give it a try myself. I still don't fully understand how these devices with just the EHCI controllers work with low and full speed. But as I have received data from a keyboard it's definitely possible.