hathach/tinyusb #2236: TU_ASSERT(epnum < EP_MAX) in dcd_synopsys.c fails

klaus-liebler, 2023-08-28:
### Operating System

Windows 11

### Board

DevEBox STM32F407

### Firmware

Created very basic test project with STM32CubeIDE and followed https://github.com/hathach/tinyusb/discussions/633 with current master branch. Used https://github.com/hathach/tinyusb/tree/master/examples/device/cdc_dual_ports as template.

Relevant/modified lines in tusb_config.h are:
```
#define CFG_TUSB_MCU OPT_MCU_STM32F4
#define CFG_TUSB_RHPORT0_MODE       (OPT_MODE_DEVICE | OPT_MODE_FULL_SPEED)
#define CFG_TUSB_RHPORT1_MODE		(OPT_MODE_NONE)
```

I deleted all directories in "/src/portable" except "st" because of multiple definitions of symbols

### What happened ?

Upon insertion of USB cable, the assertion 
```
 TU_ASSERT(epnum < EP_MAX);
```
line 626 in dcd_synopsys.c fails. EP_MAX is 4, epnum is 4 as well.

### How to reproduce ?

Compile project, insert usb cable

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Not available

### Screenshots

[image]

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2023-08-28:
STM32F407 doesn't support [cdc_dual_ports](https://github.com/hathach/tinyusb/tree/master/examples/device/cdc_dual_ports) example, as it has too few endpoints:

[image]

closed 2023-08-29 by hathach

klaus-liebler, 2023-08-30:
Thank you for pointing that out. I would not have believed that a STM32F4 would reach its limits so quickly...

BitLoose, 2023-08-30:
I've been following this thread, interested to know if I can setup TinyUSB for an STM32H7xx MCU for 192 and 384KHz audio streaming and VCom port?  Any info on this would be appreciated, thanks, Sorry if this is the wrong thread.