hathach/tinyusb #2237: rp2040 host won't work with 'tracer CH4' hub

Jajcus, 2023-08-29:
### Operating System

Linux

### Board

Raspberry Pi Pico

### Firmware

examples/host/cdc_msc_hid

### What happened ?

When hub is connected device seems stuck in a loop instead of detecting the hub connected. When disconnecting the hub it panics with 'Invalid speed'.  The same hub works correctly with Linux PC, a different hub works with the same Pico.

### How to reproduce ?

1. compile the example with: ``make BOARD=raspberry_pi_pico all LOG=2 ``
2. flash it to the Rasbperry Pi Pico
3. connect the tracer CH4  hub/card reader via a OTG atapter to the Pico (it won't work)
4. disconnect it (the pico will PANIC)

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

[debug.txt](https://github.com/hathach/tinyusb/files/12461907/debug.txt)

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-08-30:
won't fix since I don't have device in question. PR is welcome

Jajcus, 2023-08-30:
I would need at least some guide how to debug this and what could be the problem. I hoped someone can see what is wrong by looking at the logs – they are completely incomprehensible to me right now.