hathach/tinyusb #2247: 16 Bit Resolution Support?

OrhanYigitDurmaz, 2023-09-07:
### Related area

Please add 16 or 12 bit Analog Trigger Resolution

### Hardware specification

rp2040

### Is your feature request related to a problem?

I was using the arduino leonardo with Joystick library for my project, which had 16 bit res. support.  Now i shifted to the RP2040 platform, i miss that 16 bit resolution support. The arduino-pico core and Pico SDK all use 8 bit internally, which are all derived from you.

### Describe the solution you'd like

instead of uint8_t use uint16_t and make the device descriptor user conigurable. If there is a workaround for this, please let me know!

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

SpencerRightsma, 2023-11-02:
Any luck on this issue? Also need 16 bit resolution :(.

closed 2023-11-02 by OrhanYigitDurmaz

OrhanYigitDurmaz, 2023-11-05:
> Any luck on this issue? Also need 16 bit resolution :(.

Hi, sorry for the late reply. My implementation was like this:
[16BitTest.ino](https://gist.github.com/OrhanYigitDurmaz/060fef2d257459b56a2a11fd8275a68a)
Hope this helps.