hathach/tinyusb #2256: audio_4_channel_mic on STM32F411 with modified tud_audio_tx_done_post_load_cb

klaus-liebler, 2023-09-16:
### Operating System

Windows 11

### Board

BlackPill F411

### Firmware

audio_4_channel_mic example (unmodified and modified, see below)

### What happened ?

I successfully integrated the audio_4_channel_mic example in a new STM32CubeIDE project and used "plot_audio_samples.py" to test it. When I looked at the plot, I saw that the value 1 remains constant for two samples (see marking in the screenshot below). According to the source code, the value 1 should only be displayed for one sample.
[image]

I investigated further and changed the "Generate dummy data" code  to get unique sample values on all four channels 
```C
for(int i=0;i<48;i++){
  i2s_dummy_buffer[0][2*i]   = 100+i;
  i2s_dummy_buffer[0][2*i+1] = 200+i;
  i2s_dummy_buffer[1][2*i]   = 300+i;
  i2s_dummy_buffer[1][2*i+1] = 400+i;
}
```

Again I started "plot_audio_samples.py". 

Expected result: A sawtooth with continuously increasing values from 400->447

Actual result: A sawtooth with the values 400->446 and then a drop to 100 (47 of 48 values of second fifo, second channel, then the first value oder first fifo and first channel).

[image]

### How to reproduce ?

Compile and run example here [https://github.com/klaus-liebler/soundcard](https://github.com/klaus-liebler/soundcard)

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Not available

### Screenshots

See above

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2023-09-19:
Could you try #2259 ?

After fix:
[image]

klaus-liebler, 2023-09-19:
Thank you so much, @HiFiPhile . I can confirm, that #2259 fixes the problem.

Regards, Klaus

closed 2023-10-11 by HiFiPhile