hathach/tinyusb #2261: How to compile tinyusb-master -> examples with esp32

Firmware-beginner, 2023-09-22:
### Operating System

Windows 10

### Board

No board used till now

### Firmware

tinyusb-master\examples\host\cdc_msc_hid\src

### What happened ?

Aim is to build the above project. 
I'm using esp-idf 5.1 CMD. Tried to do idf.py set-target esp32s3 but getting a Cmake error. How to fix this error.

Error :
 You must set a FAMILY variable for the build (e.g.  rp2040, eps32s2,
 esp32s3).  You can do this via -DFAMILY=xxx on the cmake command line 

### How to reproduce ?

1. Open esp-idf 5.1 CMD and give the project location
2. Enter idf.py set-target esp32s3
3. Error received : 
    CMake Error at D:/tinyusb-master/hw/bsp/family_support.cmake:14 (message):
    You must set a FAMILY variable for the build (e.g.  rp2040, eps32s2,
    esp32s3).  You can do this via -DFAMILY=xxx on the cmake command line
    Call Stack (most recent call first):
    CMakeLists.txt:3 (include)

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

CMake Error at D:/tinyusb-master/hw/bsp/family_support.cmake:14 (message):
    You must set a FAMILY variable for the build (e.g.  rp2040, eps32s2,
    esp32s3).  You can do this via -DFAMILY=xxx on the cmake command line
    Call Stack (most recent call first):
    CMakeLists.txt:3 (include)

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2023-09-22:
for question, please use discussion

closed 2023-09-22 by hathach

Firmware-beginner, 2023-09-22:
I have wrote it in discussion