hathach/tinyusb #2277: examples/device/cdc_msc_freertos does not compile for stm32f439nucleo as-is

Lotharyx, 2023-10-05:
### Operating System

Linux

### Board

ST STM32F439-Nucleo

### Firmware

examples/device/cdc_msc_freertos

### What happened ?

Following output from `make BOARD=stm323f439nucleo all`

[buildlog.txt](https://github.com/hathach/tinyusb/files/12823131/buildlog.txt)

**WORKAROUND**:  Disable `maybe-uninitialized` as an error by adding to Makefile:
`CFLAGS_GCC += -Wno-error=maybe-uninitialized`

### How to reproduce ?

1. `git clone https://github.com/hathach/tinyusb`
2. `cd tinyusb/examples/device/cdc_msc_freertos`
3. `make BOARD=stm32f439nucleo get-deps`
4. `make BOARD=stm32f439nucleo all`

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

N/A as the build does not complete, no Debug Log generated

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Lotharyx, 2023-10-05:
Note: The example appears to run correctly on an STM32F4**2**9-Nucleo board, with the described workaround.

karlp, 2023-11-24:
Same issue with `host/cdc_msc_hid_freertos`  It's a freertos "error" (There's a configAssert guarding the potentially uninitialized variable) but I don't think we should have -Werror for third party chunks..

karlp, 2024-09-03:
fixed by 1a3f5f7a09d523872d8ab42fc3b432c19e8845b7

hathach, 2024-09-05:
thank karlp for retested this, could be closed now.

closed 2024-09-05 by hathach