hathach/tinyusb #2287: No CDC tud_cdc_rx_cb() callback with 64 bytes of data (or multiple of)

MWP, 2023-10-23:
### Operating System

Linux

### Board

STM32H7 Custom 

### Firmware

Custom.
Quite simple firmware using a currently days-old pull of tinyusb trunk.
Using the tinyusb provided dcd_synopsys.c

#define CFG_TUSB_MCU					OPT_MCU_STM32H7
#define CFG_TUSB_OS						OPT_OS_FREERTOS
#define BOARD_DEVICE_RHPORT_SPEED		OPT_MODE_FULL_SPEED
#define BOARD_DEVICE_RHPORT_NUM			0
#define CFG_TUSB_RHPORT1_MODE			(OPT_MODE_DEVICE | BOARD_DEVICE_RHPORT_SPEED)
# define CFG_TUD_ENDPOINT0_SIZE			64
#define CFG_TUD_CDC						2
#define CFG_TUD_MSC						0
#define CFG_TUD_HID						0
#define CFG_TUD_MIDI					0
#define CFG_TUD_VENDOR					0
// CDC FIFO size of TX and RX
#define CFG_TUD_CDC_RX_BUFSIZE			512
#define CFG_TUD_CDC_TX_BUFSIZE			512
// CDC Endpoint transfer buffer size, more is faster
#define CFG_TUD_CDC_EP_BUFSIZE			512
// MSC Buffer size of Device Mass storage
#define CFG_TUD_MSC_EP_BUFSIZE			1024

### What happened ?

When an amount of data that is NOT a multiple of 64 bytes of data are sent from the host PC to the STM32H7 device, the tud_cdc_rx_cb() callback is called with the expected data. 

When exactly 64 bytes of data (or a multiple of 64 bytes), is sent from the host PC to the STM32H7 device, the tud_cdc_rx_cb() callback is not called.
The callback is called with the missing data when more data is sent that is != 64 in size.

### How to reproduce ?

See above,

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

 DEVICE INIT ----------------------------------

USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 0C 00 00 00 00 00 
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 06 00 00 0A 00 
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 06 00 00 0A 00 
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 06 00 00 0A 00 
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 02 00 00 09 00 
  Get DeUoon USBD 8on on onon USBD S
onon
USBD 8onUonUSBD S0onon
USBD 8on on
USBD Setup Received 00 09 01 00 00 00 00 00 
  Set Configuration
  Open EP 81 with Size = 8
    Allocated 8 bytes at offset 4024  Open EP 02 with Size = 64
  Open EP 82 with Size = 64
    Allocated 64 bytes at offset 3960  Queue EP 02 with 512 bytes ...
  CDC opened
  Bi  onUSBD 0on onUSBD S on  oUSBD 0on onUSBD S on  on 

HOST PC SOFTWARE CONNECTION TO TTYACM0 ----------------

USBD Setup Received 21 22 03 00 00 00 00 00 
  CDC control request
  Queue EP 80 with 0 bytes ...
USBD Xfer Complete on EP 80 with 0 bytes
  Set Control Line State: DTR = 1, RTS = 1

USBD Setup Received 21 20 00 00 00 00 07 00 
  CDC control request
  Set Line Coding
  Queue EP 00 with 7 bytes ...
USBD Xfer Complete on EP 00 with 7 bytes
  0000:  00 C2 01 00 02 00 08                             |.......|
  CDC control complete
  Queue EP 80 with 0 bytes ...
USBD Xfer Complete on EP 80 with 0 bytes
USBD Xfer Complete on EP 02 with 16 bytes
  CDC xfer callback
  Queue EP 02 with 512 bytes ...
  Queue EP 82 with 16 bytes ...
USBD Xfer Complete on EP 82 with 16 bytes
  CDC xfer callback
USBD Xfer Complete on EP 02 with 4 bytes
  CDC xfer callback
  Queue EP 02 with 512 bytes ...
  Queue EP 82 with 20 bytes ...
USBD Xfer Complete on EP 82 with 20 bytes
  CDC xfer callback

HOST PC SENDING 10 BYTES OF DATA -----------------

USBD Xfer Complete on EP 02 with 10 bytes
  CDC xfer callback
  Queue EP 02 with 512 bytes ...
  Queue EP 82 with 10 bytes ...
USBD Xfer Complete on EP 82 with 10 bytes
  CDC xfer callback

HOST PC SENDING 63 BYTES OF DATA ----------------

USBD Xfer Complete on EP 02 with 63 bytes
  CDC xfer callback
  Queue EP 02 with 512 bytes ...
  Queue EP 82 with 63 bytes ...
USBD Xfer Complete on EP 82 with 63 bytes
  CDC xfer callback

HOST PC SENDING 65 BYTES OF DATA ------------------

USBD Xfer Complete on EP 02 with 65 bytes
  CDC xfer callback
  Queue EP 02 with 512 bytes ...
  Queue EP 82 with 65 bytes ...
USBD Xfer Complete on EP 82 with 65 bytes
  CDC xfer callback

HOST PC SENDING 64 BYTES OF DATA (NO CALL BACK), THEN 10 BYTES OF DATA -----------------

USBD Xfer Complete on EP 02 with 74 bytes
  CDC xfer callback
  Queue EP 02 with 512 bytes ...
  Queue EP 82 with 64 bytes ...
USBD Xfer Complete on EP 82 with 64 bytes
  CDC xfer callback
  Queue EP 82 with 10 bytes ...
USBD Xfer Complete on EP 82 with 10 bytes
  CDC xfer callback

--------------------------------------------------------

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

MWP, 2023-10-23:
This is a show-stopper for my device development.
If this is expected behavior, is there a known workaround?

HiFiPhile, 2023-10-23:
Hi @MWP 
It's an intended behavior,  For a stream like CDC, when an multiple bytes of endpoints size (64 bytes) is transfer e.g 256 byte, TinyUSB currently expect an zero-length data (ZLP packet) to end otherwise it will assume host is sending more data.

MWP, 2023-10-23:
Ok.
So is there a commonly used work-around for this on the TinyUSB side?
Something like a transfer time-out?
I'm not too familiar with the inner workings of TinyUSB, so i would very much appreciate some advice or a starting-point on a solution.
Thanks.

HiFiPhile, 2023-10-25:
If it's your host program you need to send a ZLP packet et the end to finish the transfer. For the moment there is no transfer timeout implemented.

MWP, 2023-11-14:
Finally back to this... setting CFG_TUD_CDC_EP_BUFSIZE=64 is indeed a workaround for the problem.
It may have some negative speed consequences, but they shouldn't be a problem for my application. 
Thankyou HiFiPhile!

closed 2023-11-14 by MWP

hathach, 2023-11-14:
I think it is no hurt to add an function to change the default behavior eventhough it is not a norm. That make tinyusb more flexible. e.g

```C
tud_cdc_wait_for_zlp(bool)
```

dakhnod, 2025-08-27:
I am experiencing the opposite behaviour. When executing `dd if=/dev/zero of=/dev/ttyACM0 bs=2049 count=1`, the callback is called multiple times. What can I change in order to have it called only one time?