hathach/tinyusb #2298: STM32F429-Disco board support

gustavowd, 2023-10-30:
### Related area

new port support

### Hardware specification

STM32F429-Disco

### Is your feature request related to a problem?

I'm trying to port TinyUSB for the STM32F429 Discovery development board.
Such board uses the rhport = 1 (USB_OTG_HS).

The pins are different from the current port for the STM32F4 family.

  __HAL_RCC_GPIOB_CLK_ENABLE();
  /**USB_OTG_HS GPIO Configuration
  PB13     ------> USB_OTG_HS_VBUS
  PB14     ------> USB_OTG_HS_DM
  PB15     ------> USB_OTG_HS_DP
  */
  GPIO_InitStruct.Pin = GPIO_PIN_13;
  GPIO_InitStruct.Mode = GPIO_MODE_INPUT;
  GPIO_InitStruct.Pull = GPIO_NOPULL;
  HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

  GPIO_InitStruct.Pin = GPIO_PIN_12|GPIO_PIN_14|GPIO_PIN_15;
  GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
  GPIO_InitStruct.Pull = GPIO_NOPULL;
  GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
  GPIO_InitStruct.Alternate = GPIO_AF12_OTG_HS_FS;
  HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

  /* Peripheral clock enable */
  __HAL_RCC_USB_OTG_HS_CLK_ENABLE();

It seems to me that the code is not prepared to use the High Speed USB port.
Can anyone help me please?

### Describe the solution you'd like

I beleive the sw -> portable -> st -> synopsys files do not support the High Speed USB port with the internal PHY for the STM32F4 family. I tried to edit this file, especially the dcd_init() function, but without success.

*------------------------------------------------------------------*/
/* Controller API
 *------------------------------------------------------------------*/
void dcd_init (uint8_t rhport)
{
  // Programming model begins in the last section of the chapter on the USB
  // peripheral in each Reference Manual.

  USB_OTG_GlobalTypeDef * usb_otg = GLOBAL_BASE(rhport);

  // No HNP/SRP (no OTG support), program timeout later.
  if ( rhport == 1 )
  {
        // Enable internal USB transceiver
	/* Select FS Embedded PHY */
	usb_otg->GUSBCFG |= USB_OTG_GUSBCFG_PHYSEL;

  /* Activate the USB Transceiver */
  usb_otg->GCCFG |= USB_OTG_GCCFG_PWRDWN;

#if defined(USB_HS_PHYC)
    // Highspeed with embedded UTMI PHYC

    // Select UTMI Interface
    usb_otg->GUSBCFG &= ~USB_OTG_GUSBCFG_ULPI_UTMI_SEL;
    usb_otg->GCCFG |= USB_OTG_GCCFG_PHYHSEN;

    // Enables control of a High Speed USB PHY
    USB_HS_PHYCInit();
#endif
  } else
  {
    // Enable internal PHY
    usb_otg->GUSBCFG |= USB_OTG_GUSBCFG_PHYSEL;

  }

  // Reset core after selecting PHY
  // Wait AHB IDLE, reset then wait until it is cleared
  while ((usb_otg->GRSTCTL & USB_OTG_GRSTCTL_AHBIDL) == 0U) {}
  usb_otg->GRSTCTL |= USB_OTG_GRSTCTL_CSRST;
  while ((usb_otg->GRSTCTL & USB_OTG_GRSTCTL_CSRST) == USB_OTG_GRSTCTL_CSRST) {}

  // Restart PHY clock
  *((volatile uint32_t *)(RHPORT_REGS_BASE + USB_OTG_PCGCCTL_BASE)) = 0;

  // Clear all interrupts
  usb_otg->GINTSTS |= usb_otg->GINTSTS;

  USB_OTG_DeviceTypeDef * dev = DEVICE_BASE(rhport);

  // If USB host misbehaves during status portion of control xfer
  // (non zero-length packet), send STALL back and discard.
  dev->DCFG |=  USB_OTG_DCFG_NZLSOHSK;

  // Required as part of core initialization.
  // TODO: How should mode mismatch be handled? It will cause
  // the core to stop working/require reset.
  usb_otg->GINTMSK |= USB_OTG_GINTMSK_OTGINT | USB_OTG_GINTMSK_MMISM;

  set_speed(rhport, TUD_OPT_HIGH_SPEED ? TUSB_SPEED_HIGH : TUSB_SPEED_FULL);

  usb_otg->GINTMSK |= USB_OTG_GINTMSK_RXFLVLM;
  /* Enable interrupts matching to the Device mode ONLY */
  usb_otg->GINTMSK |= USB_OTG_GINTMSK_USBSUSPM | USB_OTG_GINTMSK_USBRST |
                   USB_OTG_GINTMSK_ENUMDNEM | USB_OTG_GINTMSK_IEPINT |
                   USB_OTG_GINTMSK_OEPINT   | USB_OTG_GINTMSK_IISOIXFRM |
                   USB_OTG_GINTMSK_PXFRM_IISOOXFRM | USB_OTG_GINTMSK_WUIM;

  usb_otg->GINTMSK |= (USB_OTG_GINTMSK_SRQIM | USB_OTG_GINTMSK_OTGINT);

  set_turnaround(usb_otg, TUSB_SPEED_HIGH);
  usb_otg->GUSBCFG &= ~(USB_OTG_GUSBCFG_FHMOD | USB_OTG_GUSBCFG_FDMOD);
  usb_otg->GUSBCFG |= USB_OTG_GUSBCFG_FDMOD;

  dcd_disconnect(rhport);
  // Enable global interrupt
  usb_otg->GAHBCFG |= USB_OTG_GAHBCFG_GINT;
}

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

gustavowd, 2023-10-30:
I manage to get USB to start the enumerate process, but i'm getting this error:
[ 8578.520458] usb 3-4: new full-speed USB device number 46 using xhci_hcd
[ 8578.520627] usb 3-4: Device not responding to setup address.
[ 8578.728658] usb 3-4: Device not responding to setup address.
[ 8578.936488] usb 3-4: device not accepting address 46, error -71
[ 8578.936690] usb usb3-port4: unable to enumerate USB device
[ 8828.424773] usb 3-4: new full-speed USB device number 49 using xhci_hcd
[ 8828.556863] usb 3-4: device descriptor read/64, error -71
[ 8828.792769] usb 3-4: device descriptor read/64, error -71
[ 8829.028688] usb 3-4: new full-speed USB device number 50 using xhci_hcd
[ 8829.156807] usb 3-4: device descriptor read/64, error -71
[ 8829.392801] usb 3-4: device descriptor read/64, error -71
[ 8829.500853] usb usb3-port4: attempt power cycle
[ 8829.916779] usb 3-4: new full-speed USB device number 51 using xhci_hcd
[ 8829.916998] usb 3-4: Device not responding to setup address.
[ 8830.124973] usb 3-4: Device not responding to setup address.
[ 8830.336793] usb 3-4: device not accepting address 51, error -71
[ 8830.464834] usb 3-4: new full-speed USB device number 52 using xhci_hcd
[ 8830.464993] usb 3-4: Device not responding to setup address.
[ 8830.673061] usb 3-4: Device not responding to setup address.
[ 8830.880746] usb 3-4: device not accepting address 52, error -71
[ 8830.880834] usb usb3-port4: unable to enumerate USB device

HiFiPhile, 2023-10-30:
STM32F429 doesn't have internal UTMI phy, so even if the port support high speed you are limited to full speed.

gustavowd, 2023-10-30:
> STM32F429 doesn't have internal UTMI phy, so even if the port support high speed you are limited to full speed.

Yes, I know that. You can note in the linux dmesg info "new full speed device". I'm using the HS port configured to full speed.

#define BOARD_TUD_RHPORT     1
#define CFG_TUSB_RHPORT1_MODE    OPT_MODE_DEVICE
#define BOARD_TUD_MAX_SPEED   OPT_MODE_FULL_SPEED

 // Enable Device stack
#define CFG_TUD_ENABLED       1
#define  TUD_OPT_RHPORT		   1

 // Default is max speed that hardware controller could support with on-chip PHY
 #define CFG_TUD_MAX_SPEED     BOARD_TUD_MAX_SPEED

Not luck yet.

gustavowd, 2023-10-30:
github link to the project 

[https://github.com/gustavowd/STM32F429_tinnyUSB](https://github.com/gustavowd/STM32F429_tinnyUSB)

gustavowd, 2023-10-30:
The exact same code is working in the STM32F4-Discovery kit, but such kit uses the FS por. The port problem seems to be when using the HS port with the internal PHY with full speed. It just do not enumerate.

gustavowd, 2023-11-01:
I got it. The problem was in the clock. Everything is working now. Congrats for the project. It is great. :)

closed 2023-11-01 by gustavowd