hathach/tinyusb #2387: USB_EVT_READY stuck forever

gb88, 2023-12-23:
### Operating System

Windows 10

### Board

Custom board with nRF52840

### Firmware

Any bluefruit example

### What happened ?

when the board is connected to USB charger with wall adapter the program will stuck at line 1044 ( while ( !(USBD_EVENTCAUSE_READY_Msk & NRF_USBD->EVENTCAUSE) ) { }) in the file dcd_nrf5x.c 

### How to reproduce ?

Should happen in any board powered by USB with wall adapter where is no enumerated

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

no log available

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

ilgalga, 2024-10-20:
Hello, any solution about it? 
I've the same problem.

gb88, 2024-10-20:
I solved changing this file [image]

ilgalga, 2024-10-20:
Great! Thanks.
My approach has been to remove the USE_TINYUSB directive with:
#undef USE_TINYUSB in main.c

This because I do not need to use USB feature in my custom circuit.