hathach/tinyusb #2396: tuh_task() gets stuck when used for both device and host

hrvach, 2024-01-05:
### Operating System

Linux

### Board

RP2040 Pico

### Firmware

Latest github branch (commit 86c416d4c0fb38432460b3e11b08b9de76941bf5 (HEAD, tag: 0.15.0))

### What happened ?

Hi! 

First please allow me to thank you for making and maintaining this library, it's incredible!

The issue is a small one - I've noticed tusb_inited() does:

```
bool tusb_inited(void)
{
  bool ret = false;

#if CFG_TUD_ENABLED
  ret = ret || tud_inited();
#endif

#if CFG_TUH_ENABLED
  ret = ret || tuh_inited();
#endif

  return ret;
}
```

Which is essentially tuh_inited || tud_inited; it's tricky when both host and device are used simultaneously - tuh_task_ext() does:

```
void tuh_task_ext(uint32_t timeout_ms, bool in_isr)
{
  (void) in_isr; // not implemented yet

  // Skip if stack is not initialized
  if ( !tusb_inited() ) return;

  // Loop until there is no more events in the queue
  while (1)
  {
```

In this case, tusb_inited() will be true due to tud_inited() being true, but tuh_inited() can still be false. This will do while(1) and block main loop that's e.g. resetting watchdog :)

Should tuh_task_ext do if(!tuh_inited()) return; instead?

### How to reproduce ?

When using for both device and host, the tuh_task() blocks for too long initially.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

Have no easy way to fetch logs from device atm, apologies.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

jalmeroth, 2024-09-08:
Hi @hrvach,

I just stumbled over this issue… I think it got fixed here: https://github.com/hathach/tinyusb/commit/f8a5cde3c7fde62c3d6bd379fea6eaa9a629ded6

Best regards,
Jan

hrvach, 2024-09-10:
@jalmeroth thank you so much for pointing this out, awesome!