hathach/tinyusb #2398: MAX3421 as host: lost EP Rx transfer, race condition

IngHK, 2024-01-06:
### Operating System

Linux

### Board

NUCLEO-WL55JC1 (STM32WL55 with LoRaWAN, but no internal USB controller), external MAX3421 as host

### Firmware

own firmware running on STM32 based on STM32CubeWL Firmware Package v1.3.0

### What happened ?

I've maybe found a bug, but I'm not sure...
My application: STM32 with MAX3421 and TinyUSB host communicating with attached CDC device (RS232)
The issue: CDC Tx bulk transfers are lost depending on timing relationship of Rx/Tx bulk tranfers

I analysed the USB endpoint traffic with USB sniffer hardware (https://www.bugblat.com/products/minisniff/), inserted between MAX3421 and CDC device, and I see the the Rx EP transfers, but tuh_cdc_rx_cb is not everytime called.
It seems the failure appearance depending on the Rx/Tx timing relationship.

Is there still development in progress regarding MAX3421?
What can I test or what do I have to care or check about my setup STM32WL and MAX3421?

### How to reproduce ?

My current HW/SW setup is quite special...
I'll try to create an easy reproducible constellation without special hardware in the next days.
You can request any measurements you want to have from me.
I've logic analyzer, oscilloscope and USB sniffer hardware in my lab.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

There are no debug outputs, because the Rx EP transfer are lost as never happened.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

kowalewskijan, 2024-11-05:
Hi @IngHK, did you manage to solve the issue you had? I'm very curious because this is something that potentially might tell me whether is it relatively safe to use RX callbacks or should I rather use polling on RX.

hathach, 2024-11-05:
This should be fixed in the latest, please pull and try again

kowalewskijan, 2024-11-05:
I'm using v0.17.0 so I guess it should be fine? I suppose the issue can be closed to avoid confusion.

Thanks!

hathach, 2024-11-06:
should be fixed by https://github.com/hathach/tinyusb/pull/2731, but let wait for OP to confirm.

Szewcson, 2024-11-20:
It looks like the issue is still there in v0.17.0. When using tinyUSB in host mode with max3421 I observe that rx callback is lost after few (random number I think) consecutive operations over host cdc. When I add additional delay between operations it works normally.