hathach/tinyusb #2399: Host HUB hub_edpt_status_xfer double call (ASSERT)

Ryzee119, 2024-01-06:
### Operating System

Linux

### Board

OHCI

### Firmware

host/hub.c ([ae364b1460b91153cd94b4b0303eeda6419ff1d1](https://github.com/hathach/tinyusb/tree/ae364b1460b91153cd94b4b0303eeda6419ff1d1))

### What happened ?

When connecting a device to my downstream hub I hit `usbh_edpt_xfer_with_callback 843: ASSERT FAILED` everytime

```
// Attempt to transfer on a busy endpoint, sound like an race condition !
TU_ASSERT(ep_state->busy == 0);
```

Ive tracked down the issue to this area
https://github.com/hathach/tinyusb/blob/ae364b1460b91153cd94b4b0303eeda6419ff1d1/src/host/hub.c#L416C2-L441

This section calls `hub_edpt_status_xfer` but so does the `hub_clear_feature_complete_stub` function. This results in a double call to `hub_edpt_status_xfer`, consequently we hit the assert.

This patch fixes it
```
diff --git a/src/host/hub.c b/src/host/hub.c
index 32f5e0ac7..3bac18698 100644
--- a/src/host/hub.c
+++ b/src/host/hub.c
@@ -435,9 +435,12 @@ static void hub_port_get_status_complete (tuh_xfer_t* xfer)
     // Other changes are: L1 state
     // TODO clear change
 
-    // prepare for next hub status
-    // TODO continue with status_change, or maybe we can do it again with status
-    hub_edpt_status_xfer(daddr);
+    else
+    {
+      // prepare for next hub status
+      // TODO continue with status_change, or maybe we can do it again with status
+      hub_edpt_status_xfer(daddr);
+    }

```
Stress testing with device tree
```
HUB/
├─ HUB/
│  ├─ Device1
│  ├─ Device2
│  ├─ Device3
│  ├─ Device4
│  ├─ Device5
│  │  ├─ HUB
│  │  │  ├─Device6
│  │  │  ├─Device7
```

Before:
```
A device with address 1 is mounted
usbh_edpt_xfer_with_callback 843: ASSERT FAILED
```

After:
```
A device with address 1 is mounted
A device with address 2 is mounted
A device with address 3 is mounted
A device with address 4 is mounted
A device with address 5 is mounted
A device with address 6 is mounted
A device with address 7 is mounted
```

I'll raise a PR

### How to reproduce ?

Im using the OHCI backend but would expect this to happen in other backends too? 

Connect a device to a downstream hub.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

[log.txt](https://github.com/hathach/tinyusb/files/13851409/log.txt)

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Ryzee119, 2024-01-07:
on further inspection is seems an 'issue/oddity' with my hub which causes a different code path than 'normal' and is perhaps why this hasnt been noted to date so reproducing might be challenging.

This hub sets the `change.port_enable` bit interrupt, none of my others do

hathach, 2024-01-11:
fixed by #2400

closed 2024-01-11 by hathach