hathach/tinyusb #2447: tuh_hid_interface_protocol returns incorrect value

UKTailwind, 2024-02-01:
### Operating System

Windows 11

### Board

RP2040

### Firmware

V0.16.0 
NB: works correctly in 0.15.0

### What happened ?

Scenario 1:
Plug in a keyboard, tuh_hid_interface_protocol returns the correct value - 1
Then plug in a mouse,  tuh_hid_interface_protocol returns the correct value - 2
Then plug in a gamepad, tuh_hid_interface_protocol returns the value for a mouse - 2 SHOULD BE 0

Scenario 2:
Plug in a gamepad, tuh_hid_interface_protocol returns the correct value - 0
Then plug in a mouse,  tuh_hid_interface_protocol returns the correct value - 21
Then plug in a gamepad, tuh_hid_interface_protocol returns the value for a mouse - 2 SHOULD BE 0

In other words whenever a mouse or keyboard is inserted and then removed plugging a gamepad afterwards will return the last non-zero value

### How to reproduce ?

As above

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

1, 1C4F, 2
US USB Keyboard Connected on channel 1
> USB Keyboard Disconnected
> 2, 15D9, A4C
USB Mouse Connected on channel 1
> 2, 54C, 9CC
USB Mouse Connected on channel 1
>  L--

### Screenshots

N.A

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

UKTailwind, 2024-03-28:
Any update? This is a definite bug introduced after 0.15 and still there in the latest code

UKTailwind, 2024-03-29:
I've solved this but does anyone read these issues? The issue is that tinyusb re-uses an itf without clearing the protocol and does not set it if the new protocol is zero

fix:

 `static hidh_interface_t* find_new_itf(void) {
  for (uint8_t i = 0; i < CFG_TUH_HID; i++) {
    if (_hidh_itf[i].daddr == 0) {
      _hidh_itf[i].itf_protocol=0; //Clear the protocol - needed if the new protocol is zero
      return &_hidh_itf[i];
    }
  }
  return NULL;
}
`

hathach, 2024-03-31:
thank you for find the root cause, it should be fixed by 2559

PR #2559 (2024-03-31): fix hid close not clear data
**Describe the PR**
fix #2447
closed 2024-03-31 by hathach