hathach/tinyusb #2478: (RP2040) tud_connected() and tud_mounted() stay true after disconnected

acscd, 2024-02-18:
### Operating System

Linux

### Board

Raspberry Pi Pico

### Firmware

```
#include "Adafruit_TinyUSB.h"

void setup()
{
	pinMode(LED_BUILTIN, OUTPUT);
	Serial.begin(115200);
	Serial1.begin(115200); //for debug
}

void loop()
{
	digitalWrite(LED_BUILTIN, tud_connected());
}

```

### What happened ?

tud_connected() and tud_mounted() correctly return FALSE before USB is connected, and TRUE when connected. However, they continue to return TRUE indefinitely after USB is no longer connected.

### How to reproduce ?

1. power up Pi Pico from an external source (I feed VBUS thru a diode)
2. (LED stays off, tud_connected() returning FALSE)
3. Connect Pico to USB host
4. (LED turns on, tud_connected() returning TRUE)
5. Disconnect USB
6. (LED stays on, tud_connected() keeps returning true)
7. Scratch head
8. if USB is reconnected, LED will flicker off a couple of times, then stay on
9. Scratch head some more

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

[usb log.txt](https://github.com/hathach/tinyusb/files/14324629/usb.log.txt)

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

doctea, 2024-04-13:
I'm seeing the same behaviour here too.  I'm trying to figure out why sending MIDI notes to the host randomly stops working until I unplug and re-plug the USB, but as in the issue report, once connected tud_connected() and tud_mounted() continue to report that they are connected even when the USB cable is unplugged.

acscd, 2024-04-16:
On the PI Pico, depending on your use case, this issue may be worked around by sensing VBUS instead; GPIO 24 is hardwired for that on the Pico. (WL_GPIO 2 on the PicoW) This will only work if VBUS is fed exclusively by USB though.

doctea, 2024-05-04:
> On the PI Pico, depending on your use case, this issue may be worked around by sensing VBUS instead; GPIO 24 is hardwired for that on the Pico. (WL_GPIO 2 on the PicoW) This will only work if VBUS is fed exclusively by USB though.

Good to know; but in my case the +v USB pin is cut and the Pico (actually Seeed XIAO RP2040) is externally powered so this wouldn't be a suitable workaround.

HiFiPhile, 2024-05-04:
Without VBUS there is no way to do disconnection detection.

https://github.com/hathach/tinyusb/blob/a435befcdeb6bbd40cf3ba342756f8d73f031957/src/portable/raspberrypi/rp2040/dcd_rp2040.c#L44

You should use `tud_ready()`