hathach/tinyusb #2537: `tud_cdc_rx_cb` callback is never called.

jay94ks, 2024-03-24:
### Operating System

Windows 11

### Board

RP2040 custom board.

### Firmware

Full source code and full schematic can be found in [my repository](https://github.com/jay94ks/simple-np).
I'm trying to complete RP2040 based custom number-pad firmware.

**Quick link to few necessary files:**
* [USB descriptor](https://github.com/jay94ks/simple-np/blob/main/fw/board/usbd/descriptors.cpp)
* [HID report functionality](https://github.com/jay94ks/simple-np/blob/main/fw/board/usbd/hid_notifier.cpp)
* [CDC RX callback](https://github.com/jay94ks/simple-np/blob/main/fw/board/usbd.cpp)

**Board picture:**
[image]

### What happened ?

The title of this article is key subject. Callback that provides `RX` event doesn't get called. I tried almost ways I could find, but it never get called. So I concluded that this was a bug in `TinyUsb` that packaged within RP2040's C++ SDK. But that isn't clear either. The `dev_hid_composite` sample included as a sample for RP2040 appears to operate good.

### How to reproduce ?

1. Clone my repository, and compile `fw` directory then, flash it to `RP2040` development board (any kind)
2. And then, move into `app_pc`, compile `SimpleNP.Acm.TestApp` program and launch it.
3. Enter `COMxx` port to connect to the board.
4. Wire `GP0` -> `GP8` pin to press `1` key on number-pad.
5. No CDC `NOP` command executed that sent by program in 2nd stage.
6. Even more, nothing printed through TFT display (currently it works like TTY console).

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

No way to capture logs because it causes conflict with my USB descriptor. And I have no debugger except `TFT` display.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2024-03-24:
Not a TinyUSB issue, I suggest you to check linker and disassembly output to ensure your `tud_cdc_rx_cb` implementation is correctly used.

closed 2024-03-24 by HiFiPhile

jay94ks, 2024-03-24:
> Not a TinyUSB issue, I suggest you to check linker and disassembly output to ensure your `tud_cdc_rx_cb` implementation is correctly used.

`simple_np.elf.map` file have following lines:
```
 .text.tud_cdc_rx_cb
                0x10000664       0x34 CMakeFiles/simple_np.dir/board/usbd.cpp.obj
                0x10000664                tud_cdc_rx_cb
```

Isn't this correct?

HiFiPhile, 2024-03-24:
Please check the disassembly output of final binary.
https://github.com/hathach/tinyusb/blob/3ec22e24ed3e8169b7aa8bd51f86f95dda0e6002/src/class/cdc/cdc_device.c#L457

You can start with official examples on supported board, like adding following code to `cdc_dual_ports` example.
```
void tud_cdc_rx_cb(uint8_t itf) {
  static uint8_t state=0;
  board_led_write(state++);
}
```

jay94ks, 2024-03-24:
> Please check the disassembly output of final binary.
>
> https://github.com/hathach/tinyusb/blob/3ec22e24ed3e8169b7aa8bd51f86f95dda0e6002/src/class/cdc/cdc_device.c#L457
>
> You can start with official [...]

```
10000664 <tud_cdc_rx_cb>:
10000664:	b510      	push	{r4, lr}
10000666:	b090      	sub	sp, #64	; 0x40
10000668:	0004      	movs	r4, r0
1000066a:	2300      	movs	r3, #0
1000066c:	9300      	str	r3, [sp, #0]
1000066e:	223c      	movs	r2, #60	; 0x3c
10000670:	2100      	movs	r1, #0
10000672:	a801      	add	r0, sp, #4
10000674:	f006 ffc8 	bl	10007608 <__wrap_memset>
10000678:	2240      	movs	r2, #64	; 0x40
1000067a:	4669      	mov	r1, sp
1000067c:	0020      	movs	r0, r4
1000067e:	f008 fb17 	bl	10008cb0 <tud_cdc_n_read>
10000682:	1e04      	subs	r4, r0, #0
10000684:	d101      	bne.n	1000068a <tud_cdc_rx_cb+0x26>
10000686:	b010      	add	sp, #64	; 0x40
10000688:	bd10      	pop	{r4, pc}
1000068a:	f7ff ff53 	bl	10000534 <_ZN4Usbd3getEv>
1000068e:	0022      	movs	r2, r4
10000690:	4669      	mov	r1, sp
10000692:	f7ff ffbf 	bl	10000614 <_ZN4Usbd8pushRbufEPhm>
10000696:	e7f6      	b.n	10000686 <tud_cdc_rx_cb+0x22>
```

there is no reference which calls this callback `cdcd_xfer_cb` in full disassembly.
I rewrote entire codes to solve this problem, but it doesn't be solved.
And even more, basic examples are all working well if descriptor has no `HID` + `CDC` combination.

```
100090fc <cdcd_xfer_cb>:
100090fc:	b5f0      	push	{r4, r5, r6, r7, lr}
100090fe:	46d6      	mov	lr, sl
10009100:	b500      	push	{lr}
10009102:	b082      	sub	sp, #8
........................... blahblah .................
100091b2:	0038      	movs	r0, r7
100091b4:	f7f7 fa56 	bl	10000664 <>                     ** <-- No symbol presents, but callback address is correct. **
100091b8:	e7f7      	b.n	100091aa <cdcd_xfer_cb+0xae>
100091ba:	4b1a      	ldr	r3, [pc, #104]	; (10009224 <cdcd_xfer_cb+0x128>)
........................... blahblah .................
```

Full disassembled file: [simple_np.txt](https://github.com/hathach/tinyusb/files/14736102/simple_np.txt)
If the version of `tinyusb` in `pico-sdk` has corrupted, how can I replace it to other distribution?

HiFiPhile, 2024-03-24:
> there is no reference which calls this callback cdcd_xfer_cb in full disassembly.

But it's there in your attached disassembly.

> And even more, basic examples are all working well if descriptor has no HID + CDC combination.

So `cdc_dual_ports` works ? In the case of HID + CDC, does CDC data receive actually work (by `tud_cdc_n_available` and `tud_cdc_n_read` in main loop) ?

> If the version of tinyusb in pico-sdk has corrupted, how can I replace it to other distribution?

You can replace it with master branch.

jay94ks, 2024-03-24:
> > there is no reference which calls this callback cdcd_xfer_cb in full disassembly.
> 
> But it's there in your attached disassembly.
Yeah, but the instruction that calls the address of that function cannot be found.
And even more, I tested `TUD_CFG_CDC` value using `vsprintf`, and it says `usb: cdc: 1`.

```
tty_printf("usb: cdc: %d.\n", CFG_TUD_CDC);
```

> 
> > And even more, basic examples are all working well if descriptor has no HID + CDC combination.
> 
> So `cdc_dual_ports` works ? In the case of HID + CDC, does CDC data receive actually work (by `tud_cdc_n_available` and `tud_cdc_n_read` in main loop) ?

What is very frustrating is that both HID and `tud_cdc_transmit` operate perfectly, but when `tud_cdc_read_n` is called in the main loop, the USB is not recognized, and `tud_cdc_n_available(0)` always returns 0.

```
int main(void) {
    // ............................. (capsulized initialization codes) ....................
    usbd->enableHid();
    while(true) {
        kbd->scanOnce();
        led->updateOnce();
        usbd->stepOnce();

        if (tud_cdc_n_available(0)) { // --> always return zero.
            tud_cdc_rx_cb(0);
        }
    }
}
```

And, without `tud_cdc_n_available(0)` call:
 ```
int main(void) {
    // ............................. (capsulized initialization codes) ....................
    usbd->enableHid();
    while(true) {
        kbd->scanOnce();
        led->updateOnce();
        usbd->stepOnce();
        tud_cdc_rx_cb(0); // --> this hangs up here. no led toggles act after this code.
    }
}
```
It's currently 2 am where I live, so I have to go to work at morning. I'll sleep now and evening I'll capture the `cdc_dual_ports` example it's working.

> > If the version of tinyusb in pico-sdk has corrupted, how can I replace it to other distribution?
> 
> You can replace it with master branch.
Okay, I'll try this way.

jay94ks, 2024-03-26:
I updated `lib/tinyusb` of `pico-sdk` to latest release.

```
PS C:\ARM\RP2040\pico-sdk> git submodule status
 72ef1732c954d938091467961e41f4aa9b976b34 lib/btstack (v1.4-2371-g72ef1732c)
 8ef38a6d32c54f850bff8f189bdca19ded33792a lib/cyw43-driver (v1.0.1)
 239918ccc173cb2c2a62f41a40fd893f57faf1d6 lib/lwip (STABLE-2_1_0_RELEASE-548-g239918cc)
 a77287f8fa6b76f74984121fdafc8563147435c8 lib/mbedtls (v2.28.1-36-ga77287f8f)
 86c416d4c0fb38432460b3e11b08b9de76941bf5 **lib/tinyusb (0.15.0)**

...
PS C:\ARM\RP2040\pico-sdk> git submodule status
+272986f17af35a67815ab20897f6c91e710322a4 lib/btstack (v1.6)
+9f6405f0b3260968306d782e1c5ac275a46dc65d lib/cyw43-driver (v1.0.3)
+d0efd9ef7ba08e54b46b1060e2b4629a4907391b lib/lwip (STABLE-2_2_0_RELEASE-77-gd0efd9ef)
+611f899c0c9d397baedfaec34ea0861ad2543991 lib/mbedtls (v3.5.2-2690-g611f899c0)
+5b0e255f7e5f49909be9d4a0bf0be302d26e598d **lib/tinyusb (0.16.0-265-g5b0e255f7)**
```

then the RX callback is called perfectly, thanks :)
finally, solved this problem.

mathieudinel, 2025-09-15:
Hi, I am experiencing a similar issue, the callback is never called. I use a stm32g0 and a samd21 as my portable examples.

I use tinyusb as a 3rd party package to an existing private framework, and I put the glue around it myself. (I only use the /portable layer of the repo). The examples like cdc_msc and cdc_dual work fine, however when I try to put the `cdc_task` inside `tud_cdc_rx_cb`, `tud_cdc_rx_cb` is never called.

A look at disassembly tells me the weak link substitution is not occuring as expected, because the evaluation in `cdcd_xfer_cb` is optimized as 0x00000000.

Do I need to add anything special to my linker scripts to comply with this new-to-me calling scheme (weak prototypes, no definitions)? The problems occurs both with `armgcc` and `arm compiler V5 (Keil)`.