hathach/tinyusb #2554: Add a check to see if device ID is correctly mapped before accessing it

PwnVerse, 2024-03-28:
### Operating System

Linux

### Board

Feather nRF52840 Express

### Firmware

In the function `board_get_unique_id` from `hw/bsp/nrf/family.c`, here is the relevant section of the code - 

```c
size_t board_get_unique_id(uint8_t id[], size_t max_len) {
  (void) max_len;

#ifdef NRF5340_XXAA
  uintptr_t did_addr = (uintptr_t) NRF_FICR->INFO.DEVICEID;
#else
  uintptr_t did_addr = (uintptr_t) NRF_FICR->DEVICEID;
#endif

  const uint8_t* device_id = (const uint8_t*) did_addr;
  for(uint8_t i=0; i<8; i++) {
    id[i] = device_id[i];
  }
  return 8;
}
```

The device ID is read from the `NRF_FICR` and directly accessed in the subsequent for loop without any checks on whether the ID is appropriately mapped or not. This can cause `NULL` pointer or arbitrary writes in a tampered environment.

### What happened ?

I am trying to fuzz an instrumented version of the tinyUSB example application `cdc_msc_freertos`, and found a crash at the effected function (because of corrupted `NRF_FICR` register).

Here is the not so informational ASAN backtrace - 

```UndefinedBehaviorSanitizer:DEADLYSIGNAL
==2339547==ERROR: UndefinedBehaviorSanitizer: SEGV on unknown address 0x08213d38 (pc 0x0807f36b bp 0xf6291278 sp 0xf6291240 T2339549)
==2339547==The signal is caused by a WRITE memory access.
    #0 0x807f36b  (/home/ritvik/Desktop/Files/test/tinyusb_cdc_ins+0x807f36b)
    #1 0x80e0618  (/home/ritvik/Desktop/Files/test/tinyusb_cdc_ins+0x80e0618)
    #2 0x8084ace  (/home/ritvik/Desktop/Files/test/tinyusb_cdc_ins+0x8084ace)
    #3 0xf5c86c00  (/lib/i386-linux-gnu/libc.so.6+0x86c00)
    #4 0xf5d2372b  (/lib/i386-linux-gnu/libc.so.6+0x12372b)

UndefinedBehaviorSanitizer can not provide additional info.
```

### How to reproduce ?

This is a part of our ongoing research, we have instrumented the binary and got this crash.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

On further inspection in gdb - 

```
 →  0x807f36b <board_get_unique_id+1595> mov    BYTE PTR [esi], bl
    0x807f36d <board_get_unique_id+1597> jmp    0x807f387 <board_get_unique_id+1623
    0x807f36f <board_get_unique_id+1599> lea    eax, ds:0x823afdc
    0x807f375 <board_get_unique_id+1605> add    eax, 0x60
    0x807f378 <board_get_unique_id+1608> mov    DWORD PTR [esp], eax
    0x807f37b <board_get_unique_id+1611> call   0x8207130 <__sanitizer_cov_trace_pc_guard at instrumentation/afl-compiler-rt.o.c:1469>
───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────── source:/home/ritvik/De[...].c+218 ────
    213	   uintptr_t did_addr = (uintptr_t) NRF_FICR->DEVICEID;
    214	 #endif
    215	 
    216	   const uint8_t* device_id = (const uint8_t*) did_addr;
    217	   for(uint8_t i=0; i<8; i++) {
             // id=0xf7f95258  →  [...]  0x10000060 ("`"?)
 →  218	     id[i] = device_id[i];
    219	   }
    220	   return 8;
    221	 }
```

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2024-03-28:
> The device ID is read from the NRF_FICR and directly accessed in the subsequent for loop without any checks on whether the ID is appropriately mapped or not. This can cause NULL pointer or arbitrary writes in a tampered environment.

Every device register can be null in a tampered environment, it's impossible to check every registers.

PwnVerse, 2024-03-30:
What you said definitely makes sense. But a simple Null check before using the variable in my humble opinion would not cause much performance difference. Nevertheless, I also noticed that the length of the id array is being passed to the function and isn't being used which I'm assuming is 8 bytes since the code tries to access 8 bytes of id. Removing such stale variables would also be a nice thing to see in future.

HiFiPhile, 2024-03-30:
The issue you met is specific for setup, so it's up you to add specific check.

closed 2024-03-30 by HiFiPhile

hathach, 2024-03-31:
to clarify: board_get_unique_id() is not part of tinyusb source, it is just an application-scope, exists to have usb example running only. Each application will has its own implemenation . Therefore I think It shouldn't be targeted as security for tinyusb, you should target files in `src/` instead.