hathach/tinyusb #2555: Debug logging options not enabled by tusb_config.h, ep_dir_string not defined

theosib, 2024-03-28:
### Operating System

Others

### Board

Raspberry PI Pico

### Firmware

Custom firmware. You can see my whole project here: https://github.com/theosib/pico_vga/tree/main

### What happened ?

I'm trying to debug TinyUSB. I want to find out why plugging in two USB devices at once results in no devices found. You can see more detail at https://github.com/raspberrypi/pico-sdk/issues/1680.

The first thing I did was set CFG_TUSB_DEBUG to 3 in my project's tusb_config.h, which is supposed to get included by TinyUSB for various config options. But this didn't work. I think it's because src/common/tusb_debug.h doesn't have an #include directive for this. So I added a definition for CFG_TUSB_DEBUG to my CMakeLists.txt file.

This resulted in a linker error:
```
hcd_rp2040.c:(.text._hw_endpoint_init+0xf4): undefined reference to `ep_dir_string'
```
And indeed, this function isn't defined anywhere in TinyUSB.

### How to reproduce ?

- Define CFG_TUSB_DEBUG to 3 in CMakeLists.txt (`target_compile_definitions(${PROJECT} PRIVATE CFG_TUSB_DEBUG=3`)
)
- Build for RP Pico

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

This is actually the problem. I can't successfully enable debugging.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2024-03-28:
> The first thing I did was set CFG_TUSB_DEBUG to 3 in my project's tusb_config.h, which is supposed to get included by TinyUSB for various config options. But this didn't work. I think it's because src/common/tusb_debug.h doesn't have an #include directive for this.

Either pico-sdk or you messed up the includes.

> ` undefined reference to `ep_dir_string'`

Semms 6dc714b has some left ups...

theosib, 2024-03-29:
Actually, I updated the TinyUSB submodule and am using the latest code.

I'll try later to trace the includes, but in any case, setting CFG_TUSB_DEBUG in my tusb_config.h didn't do the job. Not sure why.

theosib, 2024-03-29:
I went ahead and pasted in the missing ep_dir_string, and I'm able to get debug messages. However, there is now a panic that occurs that did not occur without the debug messages.

hathach, 2024-03-29:
should be fixed, I normally only have DEBUG=2 with rp2040 since log via UART can mess up with timing when DEBUG=3.

closed 2024-03-29 by hathach

theosib, 2024-03-29:
Thanks for the quick fix!

Regarding my other bug report, however, the USB problem persists, and there is a panic that occurs when DEBUG=2 that doesn't occur when DEBUG=1 or 0.

PR #2557 (2024-03-29): fix rp2 debug build with level 3
**Describe the PR**
fix #2555