hathach/tinyusb #2600: unredistributable *.ld files under non-free license

josch, 2024-04-16:
### Operating System

Linux

### Board

n.a.

### Firmware

n.a.

### What happened ?

I'm trying to package tinyusb for inclusion in Debian. I was thus reading through all the copyright of the included files and found this:

```
**  (c)Copyright Ac6.
**  You may use this file as-is or modify it according to the needs of your
**  project. Distribution of this file (unmodified or modified) is not
**  permitted. Ac6 permit registered System Workbench for MCU users the
**  rights to distribute the assembled, compiled & linked contents of this
**  file as part of an application binary file, provided that it is built
**  using the System Workbench for MCU toolchain.
```

This statement is attached to the following files:

```
 hw/bsp/stm32f0/boards/stm32f072disco/STM32F072RBTx_FLASH.ld
 hw/bsp/stm32f1/boards/stm32f103_bluepill/STM32F103X8_FLASH.ld
 hw/bsp/stm32f1/boards/stm32f103_mini_2/STM32F103XC_FLASH.ld
 hw/bsp/stm32f2/boards/stm32f207nucleo/STM32F207ZGTx_FLASH.ld
 hw/bsp/stm32f3/boards/stm32f303disco/STM32F303VCTx_FLASH.ld
 hw/bsp/stm32f4/boards/feather_stm32f405/STM32F405RGTx_FLASH.ld
 hw/bsp/stm32f4/boards/pyboardv11/STM32F405RGTx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f401blackpill/STM32F401VCTx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f407disco/STM32F407VGTx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f411blackpill/STM32F411CEUx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f411disco/STM32F411VETx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f412disco/STM32F412ZGTx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f412nucleo/STM32F412ZGTx_FLASH.ld
 hw/bsp/stm32f7/boards/stlinkv3mini/STM32F723xE_FLASH.ld
 hw/bsp/stm32f7/boards/stm32f723disco/STM32F723xE_FLASH.ld
 hw/bsp/stm32f7/boards/stm32f746disco/STM32F746ZGTx_FLASH.ld
 hw/bsp/stm32f7/boards/stm32f746nucleo/STM32F746ZGTx_FLASH.ld
 hw/bsp/stm32f7/boards/stm32f767nucleo/STM32F767ZITx_FLASH.ld
 hw/bsp/stm32f7/boards/stm32f769disco/STM32F769ZITx_FLASH.ld
 hw/bsp/stm32h7/linker/stm32h743xx_flash.ld
 hw/bsp/stm32l0/boards/stm32l052dap52/STM32L052K8Ux_FLASH.ld
 hw/bsp/stm32l0/boards/stm32l0538disco/STM32L053C8Tx_FLASH.ld
 hw/bsp/stm32l4/boards/stm32l412nucleo/STM32L412KBUx_FLASH.ld
 hw/bsp/stm32l4/boards/stm32l476disco/STM32L476VGTx_FLASH.ld
 hw/bsp/stm32l4/boards/stm32l4r5nucleo/STM32L4RXxI_FLASH.ld
```

Is this copyright information correct? If yes, we cannot distribute it.

### How to reproduce ?

n.a.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

n.a.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2024-04-16:
It's an old issue from ST who hasn't update their license for years. 
I don't it's a good idea to make a Debian package as build system artifacts are in the source folder and in general it's not a writable directory in debian.
Otherwise you can use source only repo without examples:
https://github.com/hathach/tinyusb_src

josch, 2024-04-16:
> It's an old issue from ST who hasn't update their license for years.

Okay. Then I have to exclude those, sadly.

Another small nit I found is in `hw/bsp/stm32g0/boards/stm32g0b1nucleo/board.h` which says:

> Copyright (c) 2034, HiFiPhile

I guess that's a typo and that should say 2024?

> I don't it's a good idea to make a Debian package as build system artifacts are in the source folder

You mean they are getting placed there when building? Because they are not kept in git and i just missed them, right?

> and in general it's not a writable directory in debian.

That is correct. But I was able to build a firmware with a CMakeLists which had

     include(${TINYUSB_PATH}/tinyusb/hw/bsp/family_support.cmake)

just fine without any writes into `${TINYUSB_PATH}`. My intention is, to install the source into `/usr/src/tinyusb/` so that projects can make use of it from there.

CMake can just do an out-of-tree build and then nothing has to be written to `/usr/src/tinyusb/` or is there something I'm not considering?

>  Otherwise you can use source only repo without examples:

Thanks, but that does not include enough things. For example the cmake file referenced above would be missing.

Thank you for your quick reply!

HiFiPhile, 2024-04-16:
> Copyright (c) 2034, HiFiPhile

Haha good spot

> You mean they are getting placed there when building

Yes, also when you use batch build script `tools/build_***.py`

> CMake can just do an out-of-tree build and then nothing has to be written to /usr/src/tinyusb/

Yes if you use CMake only.

josch, 2024-04-16:
> Yes if you use CMake only.

Okay, that is what I'd like to use this for. As you point out, it doesn't make sense to package this for use-cases that require the source directory being writable. Those who need that can either `cp -a` the files from `/usr/src/tinyusb/` or `git clone` as usual.

I have another copyright question. The following files claim they are auto-generated:

```
hw/bsp/lpc11/boards/lpcxpresso11u37/lpc11u37.ld
hw/bsp/lpc11/boards/lpcxpresso11u68/lpc11u68.ld
hw/bsp/lpc13/boards/lpcxpresso1347/lpc1347.ld
hw/bsp/lpc15/boards/lpcxpresso1549/lpc1549.ld
hw/bsp/lpc17/boards/lpcxpresso1769/lpc1769.ld
hw/bsp/lpc17/boards/mbed1768/lpc1768.ld
hw/bsp/lpc18/boards/lpcxpresso18s37/lpc1837.ld
hw/bsp/lpc18/boards/mcb1800/lpc1857.ld
hw/bsp/lpc40/boards/ea4088_quickstart/lpc4088.ld
hw/bsp/lpc43/boards/ea4357/lpc4357.ld
hw/bsp/lpc43/boards/lpcxpresso43s67/lpc4367.ld
```

 - is it possible to generate those using FOSS utilities?
 - is the source they are generated from included here?
 - is anything included in them that makes these generated files non-free?
 - ultimately, what is their copyright situation, given that they were created using a proprietary tool and might contain bits that are not FOSS?

HiFiPhile, 2024-04-16:
They are generated by MCUXpresso which is not FOSS. But the SDK itself is BSD-3 licensed https://github.com/nxp-mcuxpresso/mcux-sdk/tree/main.

josch, 2024-04-16:
But the boards from my list above seem to be missing from the list at https://github.com/nxp-mcuxpresso/mcux-sdk/tree/main/boards or am I looking at the wrong place?

HiFiPhile, 2024-04-16:
They are part of older sdk, if you don't need just exclude them.

hathach, 2024-04-16:
thanks for raising the issue, I will revise and update those file with a better license, find a better alternative etc ... These are very old linker that is in 2000 era, where vendor has very strict license.

josch, 2024-04-18:
For what it's worth, here is the final copyright file in DEP5 format. Note, that not all files in the `Files-Excluded` field are interesting for you because Debian's rules are more strict than the rules that you are subject to. I omitted the copyright texts themselves for brevity.

```
Format: https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/
Upstream-Name: tinyusb
Upstream-Contact: https://github.com/hathach/tinyusb/issues/
Source: https://github.com/hathach/tinyusb
# excluded because either:
#  - auto-generated and no sources
#  - auto-generated using proprietary tool
#  - license non-free because disallows distribution
#  - license non-free because restricts usage to products of company X
Files-Excluded:
 hw/bsp/imxrt/boards/metro_m7_1011/board/pin_mux.c
 hw/bsp/imxrt/boards/metro_m7_1011/board/pin_mux.h
 hw/bsp/imxrt/boards/metro_m7_1011_sd/board/pin_mux.c
 hw/bsp/imxrt/boards/metro_m7_1011_sd/board/pin_mux.h
 hw/bsp/imxrt/boards/mimxrt1010_evk/board/pin_mux.c
 hw/bsp/imxrt/boards/mimxrt1010_evk/board/pin_mux.h
 hw/bsp/imxrt/boards/mimxrt1015_evk/board/pin_mux.c
 hw/bsp/imxrt/boards/mimxrt1015_evk/board/pin_mux.h
 hw/bsp/imxrt/boards/mimxrt1020_evk/board/pin_mux.c
 hw/bsp/imxrt/boards/mimxrt1020_evk/board/pin_mux.h
 hw/bsp/imxrt/boards/mimxrt1024_evk/board/pin_mux.c
 hw/bsp/imxrt/boards/mimxrt1024_evk/board/pin_mux.h
 hw/bsp/imxrt/boards/mimxrt1050_evkb/board/pin_mux.c
 hw/bsp/imxrt/boards/mimxrt1050_evkb/board/pin_mux.h
 hw/bsp/imxrt/boards/mimxrt1060_evk/board/pin_mux.c
 hw/bsp/imxrt/boards/mimxrt1060_evk/board/pin_mux.h
 hw/bsp/imxrt/boards/mimxrt1064_evk/board/pin_mux.c
 hw/bsp/imxrt/boards/mimxrt1064_evk/board/pin_mux.h
 hw/bsp/imxrt/boards/mimxrt1170_evkb/board/pin_mux.c
 hw/bsp/imxrt/boards/mimxrt1170_evkb/board/pin_mux.h
 hw/bsp/imxrt/boards/teensy_40/board/pin_mux.c
 hw/bsp/imxrt/boards/teensy_40/board/pin_mux.h
 hw/bsp/imxrt/boards/teensy_41/board/pin_mux.c
 hw/bsp/imxrt/boards/teensy_41/board/pin_mux.h
 hw/bsp/lpc11/boards/lpcxpresso11u37/lpc11u37.ld
 hw/bsp/lpc11/boards/lpcxpresso11u68/lpc11u68.ld
 hw/bsp/lpc13/boards/lpcxpresso1347/lpc1347.ld
 hw/bsp/lpc15/boards/lpcxpresso1549/lpc1549.ld
 hw/bsp/lpc17/boards/lpcxpresso1769/lpc1769.ld
 hw/bsp/lpc17/boards/mbed1768/lpc1768.ld
 hw/bsp/lpc18/boards/lpcxpresso18s37/lpc1837.ld
 hw/bsp/lpc18/boards/mcb1800/lpc1857.ld
 hw/bsp/lpc40/boards/ea4088_quickstart/lpc4088.ld
 hw/bsp/lpc43/boards/ea4357/lpc4357.ld
 hw/bsp/lpc43/boards/ea4357/pca9532.c
 hw/bsp/lpc43/boards/ea4357/pca9532.h
 hw/bsp/lpc43/boards/lpcxpresso43s67/lpc4367.ld
 hw/bsp/stm32f0/boards/stm32f072disco/STM32F072RBTx_FLASH.ld
 hw/bsp/stm32f1/boards/stm32f103_bluepill/STM32F103X8_FLASH.ld
 hw/bsp/stm32f1/boards/stm32f103_mini_2/STM32F103XC_FLASH.ld
 hw/bsp/stm32f2/boards/stm32f207nucleo/STM32F207ZGTx_FLASH.ld
 hw/bsp/stm32f3/boards/stm32f303disco/STM32F303VCTx_FLASH.ld
 hw/bsp/stm32f4/boards/feather_stm32f405/STM32F405RGTx_FLASH.ld
 hw/bsp/stm32f4/boards/pyboardv11/STM32F405RGTx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f401blackpill/STM32F401VCTx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f407disco/STM32F407VGTx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f411blackpill/STM32F411CEUx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f411disco/STM32F411VETx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f412disco/STM32F412ZGTx_FLASH.ld
 hw/bsp/stm32f4/boards/stm32f412nucleo/STM32F412ZGTx_FLASH.ld
 hw/bsp/stm32f7/boards/stlinkv3mini/STM32F723xE_FLASH.ld
 hw/bsp/stm32f7/boards/stm32f723disco/STM32F723xE_FLASH.ld
 hw/bsp/stm32f7/boards/stm32f746disco/STM32F746ZGTx_FLASH.ld
 hw/bsp/stm32f7/boards/stm32f746nucleo/STM32F746ZGTx_FLASH.ld
 hw/bsp/stm32f7/boards/stm32f767nucleo/STM32F767ZITx_FLASH.ld
 hw/bsp/stm32f7/boards/stm32f769disco/STM32F769ZITx_FLASH.ld
 hw/bsp/stm32h7/linker/stm32h743xx_flash.ld
[... 283 lines cut ...]
       hw/bsp/stm32f7/FreeRTOSConfig/*
       hw/bsp/stm32g0/FreeRTOSConfig/*
       hw/bsp/stm32g4/FreeRTOSConfig/*
       hw/bsp/stm32h7/FreeRTOSConfig/*
       hw/bsp/stm32l4/FreeRTOSConfig/*
       hw/bsp/stm32u5/FreeRTOSConfig/*
Copyright: 2017, Amazon.com, Inc. or its affiliates
License: MIT/X11

Files: lib/SEGGER_RTT/*
Copyright: 1995-2021, SEGGER Microcontroller GmbH
License: custom-segger

Files: lib/networking/ndis.h
Copyright: Benedikt Spranger <b.spranger@pengutronix.de>
           Casper S. Hornstrup <chorns@users.sourceforge.net>
           Colin O'Flynn <coflynn@newae.com>
License: public-domain

Files: .pre-commit-config.yaml
Copyright: 2020, Diego Elio Pettenò
License: Unlicense
```

Feel free to close this issue whenever you see fit.