hathach/tinyusb #2602: dcd_rusb2.c MSC enum error

tyustli, 2024-04-17:
### Operating System

Windows 11

### Board

ra8d1_evk

### Firmware

device cdc_msc

### What happened ?

[image]

### How to reproduce ?

add dcd_rusb.c to ra8d1 project
```c
void hal_entry(void) {
  board_init();

  // init device stack on configured roothub port
  tud_init(BOARD_TUD_RHPORT);

  if (board_init_after_tusb) {
    board_init_after_tusb();
  }

  while (1) {
    tud_task(); // tinyusb device task
    led_blinking_task();
    cdc_task();
  }
}

```

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

no debug tool

### Screenshots

[image]

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Rbb666, 2024-04-17:
This problem can also be reproduced on RA6M5/other RA series MCU platforms.

HiFiPhile, 2024-04-17:
> no debug tool

Isn't the board has on-board jlink debugger ?

tyustli, 2024-04-17:
> > no debug tool
> 
> Isn't the board has on-board jlink debugger ?

sorry I don't have jlink tool

The original capture log file cannot be uploaded either.

The picture is a screenshot of the usb packet capturer
（https://www.totalphase.com/products/control-center-serial/）

hathach, 2024-04-17:
ra8d1_ek is not supported yet in the bsp.

Rbb666, 2024-04-17:
> 这个问题也可以在RA6M5/其他RA系列MCU平台上重现。

This problem can also be reproduced on other RA series MCU platforms.

hathach, 2024-04-17:
> > 这个问题也可以在RA6M5/其他RA系列MCU平台上重现。
> 
> 
> 
> This problem can also be reproduced on other RA series MCU platforms.
> 
> 

Please provide specific board information. Make sure you use the stock example from tinyusb repo, the 1st seems to use other examples.