hathach/tinyusb #2634: Support ESP32-S3 USB Host

marchingband, 2024-05-09:
### Related area

I would like to have tinyUSB support USB Host on ESP32-S3

### Hardware specification

ESP32-S3

### Is your feature request related to a problem?

I have USB MIDI Host working under esp-idf, but I need to port my project to Arduino, and TinyUSB does not yet support ESP32-S3 host.

### Describe the solution you'd like

Add host support, and ideally a MIDI example :)

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

tore-espressif, 2024-07-08:
Hi guys, in case you are still interested in USB Host applications, we have it supported on ESP32-S2 and ESP32-S3 in esp-idf

Here's some documentation https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/usb_host.html

And some CDC examples https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/usb_host.html

marchingband, 2024-07-08:
@tore-espressif Thanks, yes I use that when I am on esp-idf. I have some projects that are on Arduino (ex. https://www.sparkfun.com/products/21307), and it would be very valuable to have USB Host available in both frameworks. Is there a brief way to explain on why this is especially challenging for the developers? The question comes up fairly frequently and I never know what to say.

niklasekstrom, 2024-09-15:
> Hi guys, in case you are still interested in USB Host applications, we have it supported on ESP32-S2 and ESP32-S3 in esp-idf

Hi Tomas,

Does your usb stack support connecting a hub, and then connecting multiple HID devices to that hub (e.g., keyboard and mouse)?

I see in this directory https://github.com/espressif/esp-idf/tree/master/components/usb that there are some files related to hubs, but I can't find any information about if hubs are actually supported.

sashahilton00, 2025-01-28:
Is there any plan to implement this in tinyusb? I'm not too familiar with the inner workings, wondering if host in tinyusb is likely or whether the only option is esp-idf for the foreseeable.

hathach, 2025-02-06:
already implemented, you can try with stock host example with _freertos or host/device_info

closed 2025-02-06 by hathach

marchingband, 2025-02-06:
@hathach are you saying it is implemented in a way that works on esp32-s3 in Arduino?

marchingband, 2025-02-07:
In the past I have used Adafruit_TinyUSB_Arduino.
Will this host controller driver be available there?
If not, what frame/lib do you recommend for using tinyusb with Arduino for ESP32-S3?
I am specifically interested in USB MIDI Host.
Thanks!

hathach, 2025-02-10:
For Arduino, it depends on arduino-esp32 core, it may start up initizliaing usb stack as device mode in which case tusb stack need to be teardown/re-initialized to host mode (api may not available yet). Sum up, To use Adafruit_TinyUSB you need to make sure
- CFG_TUH_ENABLED = 1
- usb is initialized

marchingband, 2025-02-10:
@hathach wow that is incredible news thank you!
I would love to see some Arduino examples sketches.
Is there an issue at Arduino-ESP32 I can follow to know when your updates get compiled into core?

hathach, 2025-02-11:
> [@hathach](https://github.com/hathach) wow that is incredible news thank you! I would love to see some Arduino examples sketches. Is there an issue at Arduino-ESP32 I can follow to know when your updates get compiled into core?

I haven't paid attention to arduino-esp32 lately, but It is moving fast. My CI HIL pool actually test esp32-p4 host function each PR https://github.com/hathach/tinyusb/blob/master/test/hil/tinyusb.json#L10 same dwc2 as s3 (with different usbip configuration). You can try to run stock host examples with s3, it should work.

marchingband, 2025-02-18:
That is a little beyond my skills, but I have opened an issue with core, asking for example Arduino code.
Thank you very much for your help :)
https://github.com/espressif/arduino-esp32/issues/10978

sauloverissimo, 2026-02-24:
> For anyone landing here looking for an Arduino/PlatformIO-friendly way to do USB MIDI Host on ESP32-S3, [ESP32_Host_MIDI](https://github.com/sauloverissimo/ESP32_Host_MIDI) wraps the ESP-IDF USB Host stack with a simple API. Handles MIDI enumeration, hot-plug, and multi-transport bridging (BLE, WiFi, DIN-5, etc.).