hathach/tinyusb #2650: Race condition/memory corruption on CH32V307

PoroCYon, 2024-05-18:
### Operating System

Linux

### Board

CH32V307 (using both the SCDZ devboard and a custom board)

### Firmware

Provided at https://github.com/PoroCYon/tinyusb-bug-repro --- both a CH32V307 version, and an RP2040 version which does functions correctly.

### What happened ?

On the CH32V307, using DFU, the following effects may be observed:
* The data buffer received by the DFU functions is 64 bytes off (can be in either direction)
  * This feels like a race condition thing, as it is very repeatable with `CFG_TUSB_DEBUG=2` or ASAN enabled, while without it happens much more randomly
* Random bytes in the data buffer are corrupted (watch the CRC)
* Sometimes it hangs (due to a crash due to the memory corruption, I think)
  * With payloads larger than 512 bytes, crashes seem to almost always happen.

### How to reproduce ?

Compile and load the provided firmware onto the chip. Then run `dfu-util -D payload.py`. Watch how the data in the hexdump changes, the CRC changes (for hard to spot single-byte errors), and how sometimes, hangs/crashes occur.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

[This gist](https://gist.github.com/PoroCYon/cf8b2d111263488ee6da7bf31f30c4c0) contains three log files:
* `log-vkart-rp2040.log`: RP2040, working correctly
* `log-vkart-ch32v307.log`: CH32V307, showing erratic behavior, with `CFG_TUSB_DEBUG`=**0**
* `log-vkart-ch32v307.2.log`: CH32V307, with `CFG_TUSB_DEBUG`=**2**, showing wrong but repeatable output

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

PoroCYon, 2024-05-18:
(cc @gregdavill since he is the author of the driver/backend/port?)

HiFiPhile, 2024-05-18:
There is a PR regarding ch32 issues, you can give a try: https://github.com/hathach/tinyusb/pull/2392/

PoroCYon, 2024-05-18:
#2392 indeed seems to fix the issues, thanks! (Should this issue be closed as a duplicate then?)

closed 2024-05-26 by HiFiPhile