hathach/tinyusb #2655: Settings CFG_TUD_MEM_SECTION causes a section type conflict

electretmike, 2024-05-22:
### Operating System

Linux

### Board

Custom STM32H7 based board

### Firmware

class/net/ncm_device.c

### What happened ?

When using CFG_TUD_MEM_SECTION I get a section conflict. This is because `ntb_parameters` is constant, while `transmit_ntb` isn't. See https://stackoverflow.com/questions/30076949/gcc-error-variable-causes-a-section-type-conflict

This could be fixed by adding two configuration values: `CFG_TUD_RODATA_MEM_SECTION` and `CFG_TUD_BSS_MEM_SECTION` and then replace all uses of `CFG_TUD_MEM_SECTION` with one of those two.

### How to reproduce ?

`#define CFG_TUSB_MEM_SECTION __attribute__ ((section(".usb_ram")))
`

compiler:
`arm-none-eabi-gcc (Fedora 13.2.0-5.fc40) 13.2.0
`

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

`
[build] Debug/_deps/tinyusb-src/src/class/net/ncm_device.c:140:73: error: 'ntb_parameters' causes a section type conflict with 'receive_ntb'
[build]   140 | CFG_TUD_MEM_SECTION CFG_TUSB_MEM_ALIGN tu_static const ntb_parameters_t ntb_parameters = {
[build]       |                                                                         ^~~~~~~~~~~~~~
[build] Debug/_deps/tinyusb-src/src/class/net/ncm_device.c:157:58: note: 'receive_ntb' was declared here
[build]   157 | CFG_TUD_MEM_SECTION CFG_TUSB_MEM_ALIGN tu_static uint8_t receive_ntb[CFG_TUD_NCM_OUT_NTB_MAX_SIZE];
`

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

electretmike, 2024-05-23:
Easier solution would be to remove `CFG_TUD_MEM_SECTION` from the `const` variables. Thay are not placed in RAM anyway

HiFiPhile, 2024-05-23:
Hi, thank you for the issue, The correct fix would be add an interrupt EP buffer in `CFG_TUD_MEM_SECTION`, until now the only message sent is `ntb_parameters` but it could have other messages later.

`CFG_TUD_MEM_SECTION` is necessary as explained in `tusb_config.h` that some MCUs can only transfer to/from specific RAM region.

rgrr, 2024-05-27:
Hmmm... @electretmike : which NCM are you using?  The old one or the recent one?  Recent one has my name in the copyright (Hardy Griech).

If you are at the current one: all this section stuff was already in the original NCM driver (without looking for excuses).  Thought it would be wise to not touch that part.

How is CFG_TUD_MEM_SECTION defined for your part?  Perhaps this definition is simply wrong?

hathach, 2026-11-06:
Fixed by 1533e693e: `ntb_parameters` is no longer placed in `CFG_TUD_MEM_SECTION` (`src/class/net/ncm_device.c:160`), so no const object is left in a mem section.

closed 2026-11-06 by hathach