hathach/tinyusb #2665: USB Bus Reset when running the CDC_MSC example (only CDC is enabled) on STM32U585OIYxQ custom board

DavidSayada, 2024-06-05:
### Operating System

Windows 10

### Board

Custom Board

### Firmware

This is a custom firmware (compiled using STM32CubeIDE, Version: 1.13.2, Build: 18220_20230914_1601 (UTC)), that I am sharing as a ZIP file.
[poc_tinyusb.zip](https://github.com/user-attachments/files/15579815/poc_tinyusb.zip)

### What happened ?

After the Firmware is flashed, when you connect the device to the Windows 10 host, each time the device seems enumerated, a USB bus reset is sent by the host.

### How to reproduce ?

It is enough to compile and debug the firmware using STM32CubeIDE, and then connect the device to the host.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

USBD Setup Received 80 06 00 01 00 00 40 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 0C 00 00 00 00 00
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 FF 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes ...
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 11 bytes ...
USBD Xfer Complete on EP 80 with 11 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 03 03 09 04 FF 00
  Get Descriptor String[3]
  Queue EP 80 with 34 bytes ...
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 03 00 00 FF 00
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes ...
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 FF 00
  Get Descriptor String[2]
  Queue EP 80 with 30 bytes ...
USBD Xfer Complete on EP 80 with 30 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 06 00 00 0A 00
  Get Descriptor Device Qualifier
  Stall EP0

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 03 00 00 FF 00
  Get Descriptor String[0]
  Queue EP 80 witUSBD init on controller 0, Highspeed = 0
sizeof(usbd_device_t) = 45
sizeof(dcd_event_t) = 12
sizeof(tu_fifo_t) = 12
sizeof(tu_edpt_stream_t) = 24
CDC init
guid, gsnpsid, ghwcfg1, ghwcfg2, ghwcfg3, ghwcfg4
0x00003000, 0x4F54330A, 0x00000000, 0x229ED520, 0x0200D1E8, 0x17F08030
Fullspeed PHY init
USBD Bus Reset : Full Speed

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2024-06-05:
Please use stock (stm32u575-nucleo) example for testing, your setup has many external factors.

DavidSayada, 2024-06-05:
Hi @HiFiPhile , thank you for your answer. I don't have such a board. Are there some logs I can add in the stack that may help you?

HiFiPhile, 2024-06-05:
Stock BSP only use few peripherals so most custom board should be compatible. 
https://github.com/hathach/tinyusb/blob/master/hw/bsp/stm32u5/boards/stm32u575nucleo/board.h

Try increase freertos stack size, at the end tusb reinitialized itself probably after crash reset.

DavidSayada, 2024-06-05:
Are you talking about using the hooks implemented in freertos_hook.c? If this is the case, when you say "increase freertos stack size", are you talking about the IDLE task?

DavidSayada, 2024-06-05:
The USB Bus reset is not due to a crash, but because the GINTSTS_USBRST bit is set in the DWC2 gintsts register.
For some reason the Host decides systematically to reset the USB bus.

HiFiPhile, 2024-06-05:
> increase freertos stack size", are you talking about the IDLE task?

Not the idle task, but USB task where TUSB is running.

> The USB Bus reset is not due to a crash

I'm talking about
```
USBD init on controller 0, Highspeed = 0
sizeof(usbd_device_t) = 45
sizeof(dcd_event_t) = 12
sizeof(tu_fifo_t) = 12
sizeof(tu_edpt_stream_t) = 24
``
which came only from `tusb_init()` which should only be called once in MCU init. So you are facing a crashing issue or usage fault 
as they are printed twice.

Close as it's not a TUSB issue.

closed 2024-06-05 by HiFiPhile