hathach/tinyusb #2673: USB Host: No set address recovery interval

Ryzee119, 2024-06-14:
### Operating System

Windows 11

### Board

N/A

### Firmware

N/A

### What happened ?

Ref USB Spec 9.2.6.3

> In the case of the SetAddress() request, the Status stage successfully completes when the device sends the zero-length Status packet or when the device sees the ACK in response to the Status stage data packet.
After successful completion of the Status stage, the device is allowed a SetAddress() recovery interval of 2 ms

I have a device (HID game controller) that will STALL on the  first request after the set address command is issued without this delay.

The enumeration is a bit of a complex state machine so didnt know the best way to include it. I just did this
```
     case ENUM_GET_DEVICE_DESC: {
+
+      // Wait 2ms for address recovery time, Ref USB Spec 9.2.6.3
+      osal_task_delay(2);
```

### How to reproduce ?

Very dependant on your USB device. But I  just plugged it in and got STALL error on first command after the set address. May always just

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

N/A

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2024-06-17:
this is interesting, I overlook the specs on this. Blindly wait for 2 seconds isn't good though, would your device still response after a few seconds after stalled response ? I am thinking we can allow the control transfer to retry up to 3 seconds right after SET_ADDRESS e.g. increase the delay to 1000 ms if the first descritptor isn't retrieved after addressed https://github.com/hathach/tinyusb/blob/master/src/host/usbh.c#L1314 ?

Ryzee119, 2024-06-18:
It is 2 milliseconds, not seconds so not too bad to wait on I don't think.

hathach, 2024-06-18:
ah sorry, indeed, I overlook it again, indeed 2 ms is OK :) . I think We can safely add that delay since it is part of the specs.

closed 2024-07-19 by hathach