hathach/tinyusb #2684: Vendor support outdated?

harbaum, 2024-06-20:
### Operating System

Others

### Board

Pi Pico / RP2040

### Firmware

None

### What happened ?

Compiling with 
  #define CFG_TUH_VENDOR              1
fails with 
  vendor_host.h:37:3: error: unknown type name 'pipe_handle_t'
and many errors more ...

It seems the vendor class driver has been discontinued as half of the types, constants and functions being used don't exist (anymore?).

### How to reproduce ?

Just enable TUH_VENDOR

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

No log, compilation already fails.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

harbaum, 2024-06-20:
With #2222 this driver can probably just removed at all ...

hathach, 2026-11-06:
Fixed by a57f857f8, which removed the obsolete host vendor driver; `CFG_TUH_VENDOR` no longer exists.

closed 2026-11-06 by hathach