hathach/tinyusb #2685: Add hook to allow for external host class drivers

harbaum, 2024-06-20:
### Related area

Allow developers to provide custom class drivers 

### Hardware specification

all

### Is your feature request related to a problem?

I'd like to add xbox controller support to the tuh. The xbox controllers are vendor specific and the vendor class driver seems broken: https://github.com/hathach/tinyusb/issues/2684

### Describe the solution you'd like

Allowing to add custom class drivers to  usbh_class_drivers[] in usbh.c would allow custom class drivers without the need to change/add anything to tinyusb. A simple solution may to allow the user to statically add an entry to this list via some #define in tusb_config.h

There's some discussion about this at https://github.com/hathach/tinyusb/discussions/1929. But that approach would IMHO only work if the device would report as a hid class device which the xbox controllers don't. 

There seem to be variants of tinyusb that implement something like this as this code snipped suggests:
https://github.com/RT-Thread-Studio/sdk-bsp-ra8d1-vision-board/blob/master/projects/usb/vision_board_tinyusb_xpad/board/tinyusb/host_register.c

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Ryzee119, 2024-06-20:
You've been able to hook in your own USB host drivers since https://github.com/hathach/tinyusb/pull/2222

I made this a while ago but I try keep it up to date you https://github.com/Ryzee119/tusb_xinput

harbaum, 2024-06-20:
Excellent. Your driver works as advertised ... thanks a lot ...

closed 2024-06-20 by harbaum

maaamcube, 2024-10-16:
Hello @Ryzee119 @harbaum and everyone,

I have three hubs and each hub has one USB device Vendor Specific (see image below).

[image]

then i aim to send them through wifi like this: 

[image]

Can you help with that ?
Thanks