hathach/tinyusb #2690: audio.h: error: ISO C restricts enumerator values to range of 'int' before C23 [-Werror=pedantic]

ra1nb0w, 2024-06-27:
### Operating System

Linux

### Board

STM32H563 Nucleo

### Firmware

custom

### What happened ?

These lines issue a warning/error when strictly following the ISO C standard before 23
https://github.com/hathach/tinyusb/blob/756ad3553d095b69cf3276e3967381eba958232e/src/class/audio/audio.h#L492
https://github.com/hathach/tinyusb/blob/756ad3553d095b69cf3276e3967381eba958232e/src/class/audio/audio.h#L643s

### How to reproduce ?

built with `gcc -Wfatal-errors -Wall -Wextra -Wpedantic -std=c11`

maybe a solution is to use `~0x7fffffff` as value but you know better the me the implications.
For more insight see https://stackoverflow.com/questions/11314545/casting-enum-definition-to-unsigned-int

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

```
~/dsp/libs/tinyusb/src/class/audio/audio.h:643:53: error: ISO C restricts enumerator values to range of 'int' before C23 [-Werror=pedantic]
  643 |   AUDIO_CHANNEL_CONFIG_RAW_DATA                   = 0x80000000, // TODO
      |                                                     ^~~~~~~~~~
compilation terminated due to -Wfatal-errors.
```

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

ra1nb0w, 2024-06-27:
I am also investigating another issue
```
~/libs/tinyusb/src/class/audio/audio_device.c:1493:23: error: assuming pointer wraparound does not occur when comparing P +- C1 with P +- C2 [-Werror=strict-overflow]
 1493 |         while (p_desc < p_desc_end)
      |                ~~~~~~~^~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
```

HiFiPhile, 2024-06-28:
I read the post ,does it work with `(int)(1U << 31U)` ?

ra1nb0w, 2024-06-28:
works fine. Also tested with
```
/**
 * build with
 * gcc -Wfatal-errors -Wall -Wextra -Wpedantic -std=c11
 */

#include <stdio.h>
#include <stdint.h>

#define BYTE_TO_BINARY_PATTERN "%c%c%c%c%c%c%c%c"
#define BYTE_TO_BINARY(byte)  \
  ((byte) & 0x80 ? '1' : '0'), \
  ((byte) & 0x40 ? '1' : '0'), \
  ((byte) & 0x20 ? '1' : '0'), \
  ((byte) & 0x10 ? '1' : '0'), \
  ((byte) & 0x08 ? '1' : '0'), \
  ((byte) & 0x04 ? '1' : '0'), \
  ((byte) & 0x02 ? '1' : '0'), \
  ((byte) & 0x01 ? '1' : '0') 

typedef enum
{
  A = (int)(1U << 31U),
} size_test;

int main(void)
{
  printf(BYTE_TO_BINARY_PATTERN " " BYTE_TO_BINARY_PATTERN " " BYTE_TO_BINARY_PATTERN " " BYTE_TO_BINARY_PATTERN "\n",
         BYTE_TO_BINARY(A >> 24), BYTE_TO_BINARY(A >> 16), BYTE_TO_BINARY(A >> 8), BYTE_TO_BINARY(A)
  );
}
```

Do you need a PR or do the change yourself?
Anyway, thank you very much for all your work!

ra1nb0w, 2024-07-06:
maybe we should re-open this until we find a solution

yueduz, 2024-12-27:
"1u << 30u" will fix it

HiFiPhile, 2024-12-27:
> "1u << 30u" will fix it

That's 0x40000000 not 0x80000000 !

yueduz, 2024-12-28:
I thought we didn't need such large enum values. If that's how the uac standard defines it, then we can only use #define.

yueduz, 2024-12-29:
> > "1u << 30u" will fix it
> 
> That's 0x40000000 not 0x80000000 !
The int's two's complement of "-2147483648" will be 0x80000000's complement of "uint"

hathach, 2026-11-03:
Checked while triaging hardware-free issues for #4406; left out because replacing the two `0x80000000u` enumerators is an API/ABI decision:

- Both enumerators (`src/class/audio/audio.h` ~:931, ~:964) still fail C11 `-Wpedantic -Werror`.
- Moving them to macros breaks existing C++ that assigns `RAW_DATA` to `audio20_data_format_type_I_t`.
- With GCC `-fshort-enums`, removing the high enumerator shrinks the enum from 4 bytes.

A fix needs a chosen representation that keeps the names, values, C++ use and enum size.