hathach/tinyusb #2692: tud_task() hangs when there are no more messages in the buffer due to the timeout_ms value is set to UINT32_MAX. It should be 0, right?

cornedoggen, 2024-06-28:
### Operating System

Linux

### Board

Custom

### Firmware

custom firmware

### What happened ?

when the function tud_task() is called, it will in its turn call tud_task_ext(UINT32_MAX, false). The problem with this is that if there are no more messages in the freertos queue, the timeout value of UINT32_MAX is like stay here forever.. and system hangs. I think the value of timeout could be set to 0 to immediatly return if there is no data available in the queue. 

### How to reproduce ?

try to comile a cdc example with stm32u5 and freertos configured.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

I dont have any logs. 

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

HiFiPhile, 2024-06-28:
No, please take a look at FreeRTOS enabled examples, `tud_task()` is executed in it's own task, when there is no message it's blocked to free CPU.

https://github.com/hathach/tinyusb/tree/master/examples/device/cdc_msc_freertos

closed 2024-06-28 by HiFiPhile

cornedoggen, 2024-07-22:
I still have to add a vTaskDelay after tud_task() function otherwise my application freezes. And what I ding understand is if the tud_taak is blocking, iT is consuming al cpu right?? Hoe is iT supposed to work?

HiFiPhile, 2024-07-22:
> I dont understand is if the tud_task is blocking, it is consuming all cpu right??

No, the task will enter blocking state when eveet list is empty, to give CPU resource to other tasks.

You can read more about RTOS https://www.digikey.com/en/maker/projects/introduction-to-rtos-solution-to-part-3-task-scheduling/8fbb9e0b0eed4279a2dd698f02ce125f