hathach/tinyusb #2700: tud_umount_cb not called on Raspberry Pi Pico W

Slion, 2024-07-05:
### Operating System

Others

### Board

Raspberry Pi Pico WH

### Firmware

HID device example

### What happened ?

`tud_umount_cb` is never called after calling `tud_disconnect` though if you call `tud_connect` then `tud_mount_cb` is called as expected.

### How to reproduce ?

TODO

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

TODO

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

Slion, 2024-07-06:
Updated the first post with logs and example.

HiFiPhile, 2026-11-01:
`tud_umount_cb()` is called when the stack learns the device is gone from the host. That happens in two cases: on `DCD_EVENT_UNPLUGGED`, which comes from VBUS detection, or when the host deconfigures the device with SET_CONFIGURATION(0).

`tud_disconnect()` only removes the D+ pull-up. VBUS stays present, so the device only sees the bus go idle, i.e. a suspend. It can't tell that apart from the host suspending it, so no unplug is reported.

Getting `tud_umount_cb()` on a disconnect needs VBUS detection. The RP2040 driver currently forces VBUS present (`FORCE_VBUS_DETECT` in `dcd_rp2040.c`), which is why a disconnect shows up as a suspend there.

Without VBUS detection, the application knows when it calls `tud_disconnect()` and can treat that as unmounted. Alternatively, use `tud_ready()` (mounted and not suspended), as suggested above.

Closing as not planned.

closed 2026-11-01 by HiFiPhile