hathach/tinyusb #2703: It can't work on STM32G0B1CBT6 with FreeRTOS. Please help.

favoritewky, 2024-07-10:
### Operating System

Windows 11

### Board

STM32G0B1CBT6

### Firmware

I make a simple project here https://github.com/hathach/tinyusb/commit/b3ee927e58cf7cd44f05db9dec41bc241ef94902

### What happened ?

I make a CDC Usb device but not work, I need help.
It will get 
```error: redundant redeclaration of 'SystemCoreClock' [-Werror=redundant-decls]
   53 |   extern uint32_t SystemCoreClock;
```
So, I comment this line https://github.com/hathach/tinyusb/commit/b3ee927e58cf7cd44f05db9dec41bc241ef94902#r144079217

### How to reproduce ?

Download the firmware but nothing happend.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

```
.USBD init on controller 0, Highspeed = 0
sizeof(usbd_device_t) = 53
sizeof(dcd_event_t) = 12
sizeof(tu_fifo_t) = 20
sizeof(tu_edpt_stream_t) = 112
CDC init
```

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

favoritewky, 2024-07-10:
@Commotari Please help me.

HiFiPhile, 2024-07-10:
Please don't use issue for asking for help.

closed 2024-07-10 by HiFiPhile