hathach/tinyusb #2753: MSC inquery request causes USBD Bus Reset

luftaquila, 2024-08-07:
### Operating System

Windows 11

### Board

STM32F103C8T6 Blue Pill

### Firmware

examples/device/cdc_msc

### What happened ?

When put the USB to the host, `USB Mass Storage` device appears. But it refreshes few times, then marked error.

As shown in the log output below, there are always 6 times MSC Inquery requests. And it always immediately triggers the `USBD Bus Reset`.

Device Manager's `Bus reported device description` shows the correct desciption.

### How to reproduce ?

No OS, just call `tusb_init()` after all HAL init finishes. Then, call `tud_task()` in infinite loop.

### Debug Log as txt file (LOG/CFG_TUSB_DEBUG=2)

<details>

<summary>log output</summary>

```
[03:33:51.749] tio v2.7
[03:33:51.750] Press ctrl-t q to quit
[03:33:51.796] Connected
USBD Suspend : Remote Wakeup = 0
USBD init on controller 0, Highspeed = 0
sizeof(usbd_device_t) = 52
sizeof(dcd_event_t) = 12
sizeof(tu_fifo_t) = 12
sizeof(tu_edpt_stream_t) = 24
CDC init
MSC init
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 69 00 00 00 00 00 
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 FF 00 
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes ...
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 34 bytes ...
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with 0 bytes ...
[... 462 lines cut ...]

USBD Setup Received 00 09 01 00 00 00 00 00
  Set Configuration
  Open EP 81 with Size = 8
  Open EP 02 with Size = 64
  Open EP 82 with Size = 64
  Queue EP 02 with 64 bytes ...
  CDC opened
  Bind EP 81 to driver id 0
  Bind EP 02 to driver id 0
  Bind EP 82 to driver id 0
  Open EP 03 with Size = 64
  Open EP 83 with Size = 64
  Queue EP 03 with 31 bytes ...
  MSC opened
  Bind EP 03 to driver id 1
  Bind EP 83 to driver id 1
  Queue EP 80 with 0 bytes ...
USBD Xfer Complete on EP 80 with 0 bytes
```
</details>

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

hathach, 2024-08-08:
try to see if https://github.com/hathach/tinyusb/pull/2750 fix your issue

luftaquila, 2024-08-08:
Well, I checked out to the `fix-ch32v203-setup` and builded it, now it seems to be fail at the earlier stage of the initialization.

Now Device Manager shows `Unknown USB Device (Device descriptor request failed)`, and TinyUSB log output is way shorter:

<details>

<summary>log output</summary>

```
[22:18:25.356] tio v2.7
[22:18:25.357] Press ctrl-t q to quit
[22:18:25.402] Connected
USBD init on controller 0, Highspeed = 0
sizeof(usbd_device_t) = 52
sizeof(dcd_event_t) = 12
sizeof(tu_fifo_t) = 12
sizeof(tu_edpt_stream_t) = 24
CDC init
MSC init
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 45 00 00 00 00 00 
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 46 00 00 00 00 00 
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 47 00 00 00 00 00 
[... 157 lines cut ...]

USBD Setup Received 80 06 00 01 00 00 40 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 4F 00 00 00 00 00
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Suspend : Remote Wakeup = 0
```

</details>

Thank you for your support!

luftaquila, 2024-08-08:
The most recent master that have merged `fix-ch32v203-setup` still presents the issue

HiFiPhile, 2024-09-28:
I don't have the Blue Pill but an ancient minimal board (Rev.Y chip).
Trying to make CDC and MSC requests concurrently but unable to reproduce it.

Your log looks like having some communication issue, could you verify your cabling, or a bad unit with soldering issue ?

luftaquila, 2024-09-30:
There was no issue with eye inspection on the board and the cable used worked fine in communication with other boards.

HiFiPhile, 2024-10-01:
Could you please test with `cdc_dual_ports` example and make traffic between 2 ports using something like [hterm](https://www.der-hammer.info/pages/terminal.html) ? To see if there is a signaling issue.

luftaquila, 2024-10-01:
When I tried `cdc_dual_ports` on STM32F103 Blue Pill, the Windows Device Manager recognizes both ports as `TinyUSB CDC` ports, but only one of them was able to open. It seems like that the example should echo the input, but nothing had shown on the serial monitor.

By the way, I tried `cdc_msc` example on STM32F401 Black Pill and both MSC and CDC classes worked well. But when I tried `cdc_dual_ports` on STM32F401, Windows devmgmt shows error on USB Composite Device and none of the ports are appearing.

Now I have no idea about which one is the problem, but I think I'll move to stm32f401 to use msc&cdc composite project.

luftaquila, 2024-10-02:
I don't think so, because I program my MCU several times a day and always happens same. I'll try more and tell you if I figure out something more. Thank you very much for your help.