hathach/tinyusb #2791: Upstream ESP32-P4 support

tannewt, 2024-09-06:
Espressif's TinyUSB has support for the ESP32-P4 in their release 0.15: https://github.com/espressif/tinyusb/commits/release/v0.15/ The changes are from @tore-espressif, @lijunru-hub, @roma-jam and @peter-marcisovsky.

It'd be wonderful to get these changes upstreamed so we can use them in CircuitPython alongside the other stuff that is already in upstream.

tore-espressif, 2024-09-09:
Hi,
Thank you for bringing this up. We certainly plan to upstream P4 support, we only got a little delayed with testing.
The new High-Speed support for P4 introduces many changes also for S2 and S3 with high regression risk, so we want to be sure we don't break anything.
@hathach Is there any timeline for next TinyUSB release? It would be nice to include this feature in it :)

HiFiPhile, 2024-09-09:
@tore-espressif 
Hi, is there a planning to migrate to the generic dwc2 driver instead of dcd_esp32sx ?

tore-espressif, 2024-09-09:
> Hi, is there a planning to migrate to the generic dwc2 driver instead of dcd_esp32sx ?

@HiFiPhile this is done in our [fork of TinyUSB](https://github.com/espressif/tinyusb/blob/release/v0.15/CMakeLists.txt#L49). Adding the high-speed support was a lot easier this way, as HS is already supported on STMs and other platforms. This is the huge, risky change I had in mind (most of bugs that we encountered are coming from the DCD layer)

We also wanted to evaluate changes from https://github.com/hathach/tinyusb/pull/2576 , as having DMA significantly improves performance

hathach, 2024-09-09:
@tore-espressif next release is coming very soon, I actually want to release it for quite some time but got busy with lots of other works. I think we can make a 0.17.0 this week or so. And then when P4 got merged, we can make an additional release e.g 0.17.1, that way if there is a serious bug with P4, user can easily roll back. 

PS: Thanks to @ladyada I will have the P4 eval board in a week, and can start to help testing and get it upstream.

tannewt, 2024-09-09:
Thanks! Glad to hear you are willing to upstream it. We'll hold on P4 support in CircuitPython until support is upstreamed. We don't need it in a release though.

tannewt, 2024-09-11:
> Just for completeness: ESP32-P4 has two USB peripherals, one FS and one HS, each with its dedicated PHY. So theoretically, one can work as Host and the other as Device (or both device, or both host).
> 
> As of now, we only have support for the HS peripheral. FS peripheral support is still work in progress

We'd love to do both! Device on high speed and full speed host. Support for HS first would be good so we can have device support in CircuitPython.

hathach, 2024-09-13:
> Just for completeness: ESP32-P4 has two USB peripherals, one FS and one HS, each with its dedicated PHY. So theoretically, one can work as Host and the other as Device (or both device, or both [...]

That is great to hear, yeah, @tore-espressif I just made an new release for 0.17.0. And will receive my P4 eval board real soon (in a day or two). Would you mind making an PR for P4. It does not need to be clean/perfect, I can help to test/review/work on the PR. If you are busy, I could also checkoout your branch and make a PR as well (so that your commit log is reserved) and start to work from that.

Once done we can make another 0.17.1 release or so, that would make user switch between the merge in case there is an issue with the new driver.

PS: I guess it is time to review and work on host driver for dwc2.

lijunru-hub, 2024-09-14:
I would like to know how to run two TinyUSB stacks on the P4, creating two USB devices that each operate through different USB ports.

HiFiPhile, 2024-09-14:
> I would like to know how to run two TinyUSB stacks on the P4, creating two USB devices that each operate through different USB ports.

Currently it's not possible, some works have been done with the port number but multi port simultaneously is not implemented yet.
`bool tud_init(uint8_t rhport) {`

> PS: I guess it is time to review and work on host driver for dwc2.

I hope dwc2 HCD will be much simpler than DCD, at least no need to deal with fifo allocation ;)

tore-espressif, 2024-09-18:
> Would you mind making an PR for P4?

Yes, sure. Our current timeline is roughly like this.
1. Beginning of October: Rebase our changes on release v0.17 and retest
2. Mid/End October: Open PRs with P4 support

> I hope dwc2 HCD will be much simpler than DCD, at least no need to deal with fifo allocation ;)

This will greatly depend on which memory mode you choose. There is Slave, Buffer DMA and Scatter-Gather DMA.
But my guess is that you'll go with Buffer DMA, because the Slave mode loads the CPU too much and Scatter-Gather doesn't support SPLIT transfers in HighSpeed, right?

hathach, 2024-09-23:
@tore-espressif thank you for the timeline, it is a bit long. I got the P4 eval kit just now. is in the middle of reviewing the DMA PR, once that merged. I will then move on to try to get your code for P4 specific based on your fork. So that folks at adafruit can start to do circuitpython port on it.

closed 2024-10-01 by hathach