hathach/tinyusb #2816: STM32H7R/H7S support

EamonBrennanDRS, 2024-09-25:
### Related area

New port support.

### Hardware specification

STM32H7R/STM32H7S

### Is your feature request related to a problem?

While TinyUSB already supports the H7 series, it uses an off-chip PHY over ULPI for the HS port. The H7RS series has a dedicated on-chip HS PHY, making it similar to the U5 series. The H7RS series is also distinct from the remaining H7 series, with different pin ports, and they even have their own separate HAL file, "stm32h7rsxx_hal.h".

### Describe the solution you'd like

Support for STM32H7RS series if possible.

### I have checked existing issues, dicussion and documentation

- [X] I confirm I have checked existing issues, dicussion and documentation.

EamonBrennanDRS, 2024-09-25:
Based on the reference manual ([RM0477](https://www.st.com/resource/en/reference_manual/rm0477-stm32h7rx7sx-armbased-32bit-mcus-stmicroelectronics.pdf), 3791 pages long), the OTG_FS FIFO has 1.25 Kbytes of dedicated RAM, while the OTG_HS FIFO has 4 Kbytes of dedicated RAM. Use "find in page" and search for "The USB system features" to find this.

chintal, 2024-10-31:
I would also like to see STM32H7RS support added to TinyUSB. 

I've just started trying out the H7S3L8 on its Nucleo. I might try out tinyusb on this chip in some time, if I do I will report any results. The UART is good enough for what I need to do immediately, though, so it might take a little while. 

Here's some possibly useful information:
- The chips seem to have two USBs, a USB FS and a USB HS, both with on-chip PHYs and both OTGs
- Both the FS and HS sections of the datasheet are marked as copyright Synopsys. I don't know if they are dwc2, but from other discussions I've seen in the past, it seems that's what they all are.
- The FS peripheral supports 12 host channels as host, and as device 1 bidirectional control endpoint0 and 5 IN and 5 OUT endpoints. As mentioned by OP, it has 1.25KB of dedicated RAM with FIFO controls.
- The HS peripheral supports 16 host channels as host, and as device 1 bidirectional control endpoint0 and 8 IN and 8 OUT endpoints. As mentioned by OP, it has 4KB of dedicated RAM with FIFO control.
- The CMSIS headers for the chips define `STM32H7RS`, unlike other STM32H7s which define `STM32H7`. The headers for these chips do not define `STM32H7` at all. I do not know if there is any difference in the USB peripherals, but presumably there are significant enough differences somewhere to warrant a separate family.

On a side note, the chips also have USB PD support - a UCPD peripheral can control CC1, CC2, and dead battery indication. The Nucleo also has a separate chip (TCPP03) which seems to do about the same thing. I have not worked with USB-C PD before, I don't intend to now, and I do not know if this is within the domain of what tinyUSB handles.

chintal, 2024-11-18:
@EamonBrennanDRS STM32H7RS support has been merged into master. Could you check if it works for you?

hathach, 2024-11-18:
fixed by #2871